zephyrproject-rtos/zephyr #7868: Support non-recursive builds for single-toolchain multi-core builds

SebastianBoe, 2018-05-24:
AMP architecture's like lpcxpresso54114 require multiple images to be built, one for each core. Currently this is supported in Zephyr through recursive invocations of CMake. But recursive invocation's should be avoided whenever possible.

A limitation in CMake prevents us from achieving a multi-toolchain build, but for ARM platforms like lpcxpresso54114, one toolchain is enough. Recursive-CMake is employed today because of a limitation 
inherent in the Zephyr build system, not in CMake. CMake supports building multiple application's just fine.

To avoid recursive invocations of CMake this limitation should be resolved.

NB: Future work can look at multi-toolchain builds, Kitware has expressed that this is near-unachievable. So there is hope. An interesting approach would be to evaluate if one could have a single toolchain that acted as a compiler-frontend to the multiple toolchains. Thereby avoiding multiple toolchains as far as CMake is concerned.

carlescufi, 2018-06-27:
This issue should be generalized for multi-image (i.e. single core with bootloader)

SebastianBoe, 2018-06-28:
For the record I believe there are two ways to do a non-recursive build with a bootloader.

Do a completely vanilla build and treat the bootloader as any other library that the linker happens to be very particular about how is placed.

Do a multi-image (multi-app/ multi-executable whatever you want to call it) build and merge the hex files.

(Finally, you want to support merging in a fixed hex file for the use-case where you are developing for an app that is out in the wild and therefore requires the bootloader to remain unchanged.)

carlescufi, 2020-03-17:
This will be addressed in a more generic way in the future.

closed 2020-03-17 by carlescufi