zephyrproject-rtos/zephyr #7869: Improve Zero Latency IRQ on ARM

joerchan, 2018-05-24:
The description of Zero Latency IRQs is not completely synced with the implementation.
And I don't think it quite fulfills the zero latency requirement the way it is currently implemented.

> CONFIG_ZERO_LATENCY_IRQS:
> Interrupt locking is done by setting exception masking to priority one, thus allowing exception of priority zero to still come in.
> By default, the kernel verifies, via __ASSERT() statements, that the interrupt priority is not set to zero when either connecting them setting their priority.
> Enabling this option disables the check, thus allowing setting the priority of interrupts to zero.

The description says it always disables the check and allowing priority zero, but instead it always sets priority one, and then interrupt locking is set to masking priority two and higher.
Having exception be able to take priority over zero latency IRQs is acceptable in my opinion because that would be a serious error and probably the system needs to be restarted.
However SVC is also set at priority zero, which is not good for the zero latency interrupt requirement.

SVCs can be used for three different things:

1. IRQ offloading. Cause a function to run immediately in IRQ context.
  - This I don't think is necessary to have at priority zero at all, the Config option says this is mainly used for testing.

2. Kernel panic/oops
  - Since Zero latency cannot call kernel functions this should not need to preempt the Zero latency IRQ.
    Doing kernel panic inside of exception handling is not necessary in my opinion.

3. Kernel and User mode separation by calling kernel functions through SVC trap.
  - Since Zero latency cannot call kernel functions this should not be needed to have higher priority.
    This would mean that SVC calls by the kernel supports being preempted by higher priority interrupts.

I would therefore like to propose making the following change:

When Zero Latency is enabled:
Prio 0: Exceptions
Prio 1: Zero Latency Interrupt
Prio 2: SVC
IRQ locking by masking prio 3 and higher.

Alternatively to save interrupt levels:
Prio 0: Exceptions, Zero Latency Interrupt
Prio 1: SVC
IRQ locking by masking prio 2 and higher.

When Zero latency is disabled we keep the current behavior:
Prio 0: Exceptions, SVC
IRQ locking by masking prio 1 and higher.

Update the description of Zero Latency IRQ to describe that the priority level is fixed, and that the kernel offsets the IRQ priority instead of asserting

ioannisg, 2018-05-24:
Related closed issues: to #2897, #2543

ioannisg, 2018-05-24:
CC @agross-linaro , @andrewboie , @cvinayak

PR #8109 (2018-06-20): arch: arm: Set Zero Latency IRQ to priority level zero:
Set Zero Latency IRQ to priority level zero and SVCs to priority level
one when Zero Latency IRQ is enabled.
This makes Zero Zatency truly zero latency when the kernel has been
configured with userspace enabled, or when IRQ offloading is used.

Exceptions can still delay Zero Latency IRQ, but this is considered
ok since exceptions indicate a serious error, and the system needs to
recover.

Fixes: #7869
commit 5204fd70 (2018-06-20): arch: arm: Set Zero Latency IRQ to priority level zero:

Set Zero Latency IRQ to priority level zero and SVCs to priority level
one when Zero Latency IRQ is enabled.
This makes Zero Zatency truly zero latency when the kernel has been
configured with userspace enabled, or when IRQ offloading is used.

Exceptions can still delay Zero Latency IRQ, but this is considered
ok since exceptions indicate a serious error, and the system needs to
recover.

Fixes: #7869
closed 2018-06-20 by carlescufi