zephyrproject-rtos/zephyr #7873: cc2650_sensortag_defconfig assigns missing Kconfig symbols

ulfalizer, 2018-05-24:
The following Kconfig symbols never existed, but are assigned in `boards/arm/cc2650_sensortag/cc2650_sensortag_defconfig`:

 - PINMUX_CC2650_INIT_PRIO
 - CC2650_TRNG_IRQ_PRIO
 - CC2650_TRNG_ENTROPY_GENERATOR

Only the last one is referenced elsewhere, giving some dead code:

```
arch/arm/soc/ti_simplelink/cc2650/soc.c:#ifdef CONFIG_CC2650_TRNG_ENTROPY_GENERATOR
arch/arm/soc/ti_simplelink/cc2650/soc.c:	defined(CONFIG_CC2650_TRNG_ENTROPY_GENERATOR) || \
arch/arm/soc/ti_simplelink/cc2650/soc.c:#if defined(CONFIG_GPIO_CC2650) || defined(CONFIG_CC2650_TRNG_ENTROPY_GENERATOR)
arch/arm/soc/ti_simplelink/cc2650/soc.c:#ifdef CONFIG_CC2650_TRNG_ENTROPY_GENERATOR
arch/arm/soc/ti_simplelink/cc2650/soc.c:#if defined(CONFIG_GPIO_CC2650) || defined(CONFIG_CC2650_TRNG_ENTROPY_GENERATOR)
```

The `#ifdef`s were renamed by commit da9b0ddf5beb (drivers: Rename `random` to `entropy`), but no symbol with the old name exists either.

ulfalizer, 2018-05-24:
CC @ClumsyApe @nashif 

Trying to get rid of assignments to undefined Kconfig symbols, to turn it into an error.

GAnthony, 2018-05-25:
I could remove what appears to be the dead code, but I don't have this board to test.  

@ClumsyApe, I see cc2650_sensortag doesn't build for the blinky sample, and wonder if this board is even being used.  Can you please comment on the intention of the TRNG support?

ulfalizer, 2018-05-25:
I submitted a PR that just removes the Kconfig assignments: https://github.com/zephyrproject-rtos/zephyr/pull/7929. Trying to get of all assignments to undefined symbols to turn it into an error.

Remove the dead code too if it isn't used.

PR #8036 (2018-05-31): arch: arm: soc: cc2650: Remove dead code
Remove the dead code that was guarded by previously removed
Kconfig symbols for the sensortag board.

With b028a5154265d9e68948491926383cecef78d353, this
fixes #7873 .
commit ec7d483b (2018-05-31): arch: arm: soc: cc2650: Remove dead code.

Remove the dead code that was guarded by previously removed
Kconfig symbols for the sensortag board.

With b028a5154265d9e68948491926383cecef78d353,
this fixes #7873
closed 2018-05-31 by nashif