zephyrproject-rtos/zephyr #7877: tests: kernel/mem_protect/mem_protect is failed on nucleo_f429zi/disco_l475_iot1

erwango, 2018-05-24:
Tested on e9174c6ddeddf58d8aebba5c89e50965731899e6.

andrewboie, 2018-05-24:
I don't have this hardware available to me but I did a build and examined the binary.

It appears that the MPU hardware is not being programmed correctly for the memory domain configuration. The tests are trying to read/write a buffer that they should have been granted access to but instead we fault.

I don't know if this is a recent regression, or a problem that has been around for a long time only exposed by recently enabling many more tests to run in user mode on non-x86.

I'd be very interested to know if this is limited to just this board/SOC or if all ARM MCUs that have the ARM MPU (not the NXP) are affected.

MaureenHelm, 2018-05-24:
This test passes on mimxrt1050_evk, which has the ARM MPU

andrewboie, 2018-05-24:
@erwango @nashif if you have boards can you try to run this one with assertions enabled?
Some of our SOCs define way too many static MPU regions at boot, leaving little or none left behind for runtime programming. So it may be trying to program the MPU and failing to get a free region.

andrewboie, 2018-05-24:
This is basically an instance of #6896

andrewboie, 2018-05-24:
per @nashif  #7893 fixes this

closed 2018-05-24 by andrewboie

erwango, 2018-05-25:
Fixed by #7893

PR #7893 (2018-05-24): arm: st_stm32: reduce boot MPU regions
The PPB and Peripheral regions have been removed,
the Shareable options are only important with unlocked
dual cores. Just use the background mapping.

Related to #6896 and #7877

fixes #7877