zephyrproject-rtos/zephyr #7882: tests/dfu/mcuboot.test_bank_erase fails on nrf52840_pca10056

inakypg, 2018-05-24:
commit 5b20350a722d289b85cfbaa530bdc7de377141fb 

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.0-rc1 (delayed boot 1000ms) *****
Running test suite test_mcuboot_interface
===================================================================
starting test - test_bank_erase

    Assertion failed at tests/subsys/dfu/mcuboot/src/main.c:59: test_bank_erase: (pass)
fail
FAIL - test_bank_erase
===================================================================
starting test - test_request_upgrade
PASS - test_request_upgrade
===================================================================
starting test - test_write_confirm
PASS - test_write_confirm
===================================================================
===================================================================
RunID: ci-180524-0516-1661:asla
PROJECT EXECUTION FAILED
```

inakypg, 2018-05-24:
First build this popped up in our radar was around 05/23 1609 (commit d73e970084c5e8a0f1ded28af42c64938d72341d), but this is not necessarily the pinpoint.

nvlsianpu, 2018-05-25:
thx, will looks on that

nvlsianpu, 2018-05-28:
@inakypg I can not reproduce this with current master  (SHA 71ce9b524ab9baf41096f29c5d261c6aec43354b). Can you test it as well?

nashif, 2018-05-28:
can't reproduce as well:

```
DEVICE: ***** Booting Zephyr OS v1.12.0-rc1-126-g5b20350a72 *****
DEVICE: Running test suite test_mcuboot_interface
DEVICE: ===================================================================
DEVICE: starting test - test_bank_erase
DEVICE: PASS - test_bank_erase
DEVICE: ===================================================================
DEVICE: starting test - test_request_upgrade
DEVICE: PASS - test_request_upgrade
DEVICE: ===================================================================
DEVICE: starting test - test_write_confirm
DEVICE: PASS - test_write_confirm
DEVICE: ===================================================================
DEVICE: ===================================================================
DEVICE: PROJECT EXECUTION SUCCESSFUL
```

nashif, 2018-05-28:
please reopen if you still can reproduce this.

closed 2018-05-28 by nashif

inakypg, 2018-06-28:
I keep seeing this failing consistently in two different boards that are connected to two different test servers on a different continent each, I can't pencil this no longer to transient HW issues.

@nvlsianpu what can we do to help diagnose this remotely?

nvlsianpu, 2018-06-28:
nRF52840 flash erase cycle per page endurance is 10k.

inakypg, 2018-08-29:
Thanks for your feedback -- I had missed it among the rest of the grind

This is very puzzling; I looked at all the data I have -- there are 243 confirmed failures on different pieces of hardware, all the same assertion.

It started happening on May 23rd 18 and has regularly happened until now. We've had the NRF's on the HW testing pool since before.

Once thing before I forget, that I keep forgetting, is our test infra sets CONFIG_BOOT_DELAY to 1000 (one second). Sometimes this uncovers issues.

What extra debugging can we add to get to the end of this?