zephyrproject-rtos/zephyr #7885: em_starterkit_em7d_v22 failing multiple tests in user mode

andrewboie, 2018-05-24:
We recently enabled user mode by default when running tests cases on hardware that is capable of it, and I am seeing a bunch of failures on this ARC board. The following tests are either hanging or crashing with an exception:

tests/kernel/msgq/msgq_api
tests/kernel/mem_pool/sys_mem_pool
tests/kernel/queue/
tests/kernel/mem_protect/mem_protect
tests/kernel/poll
tests/kernel/semaphore/semaphore
tests/kernel/stack/stack_api

nashif, 2018-05-24:
@vonhust  FYI

andrewboie, 2018-05-24:
Possible relationship to #7776

ruuddw, 2018-05-25:
PRs #7778 and #7923 fix most of the issues.
Remaining:
1. tests/kernel/mem_protect/mem_protect : alignment problem
    see issue #7776; @andrewboie can you have a look and suggest how best to solve? remove the 'packed' attribute?

2. tests/kernel/fatal: stack protection exception triggers, but is not reported as a STACK_CHK_FAIL (similar to ARM). I didn't investigate in detail yet.

3. tests/kernel/profiline/profiling_api: ARC combines privileged and user stack for kernel threads, test should be updated. See issue #7858

andrewboie, 2018-05-26:
> tests/kernel/fatal: stack protection exception triggers, but is not reported as a STACK_CHK_FAIL (similar to ARM). I didn't investigate in detail yet.

Hmm. This used to work, at least for the native stack overflow protection on ARC. Do we still update those registers properly?
I remember someone also added support for guard regions for the supervisor-mode stack, in that case may need to check the faulting memory address and see if it falls within bounds if the CPU doesn't report a special error when it tries to push onto the stack and fails.

andrewboie, 2018-05-30:
If the buffer is intended to be inacessible other than to the memory domain, we could also tag it with __kernel and it will not be in the app memory space.

vonhust, 2018-05-31:
most of issues left are caused by STACK_CHECKING of ARC. This is not well implemented. I am trying to provide a fix, and hope to catch up with 1.12. If not, the simplest way  is to stack check off when CONFIG_USERSPACE is on.

PR #8091 (2018-06-04): arch: arc: refactor the arc stack check support
* the original stack check codes have no consideration
  for userspace case. This will wrong cause possible stack
  check exception.
* this commit refactors the arc stack check support to
  support the usperspace.

* this commit fixes #7885. All the failed tests in #7885
  are run again to verify this commit. The test results are ok
commit b829bc78 (2018-06-04): arch: arc: refactor the arc stack check support

* the original stack check codes have no consideration
  for userspace case. This will wrong cause possible stack
  check exception.
* this commit refactors the arc stack check support to
  support the usperspace.

* this commit fixes #7885. All the failed tests in #7885
  are run again to verify this commit. The test results are ok
closed 2018-06-04 by nashif