zephyrproject-rtos/zephyr #7891: tests/posix/timer.test_timer fails on nrf51_pca10028

inakypg, 2018-05-24:
commit 4e8438bc0e1f856765f4e556d66685e65607388e

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.0-rc1 (delayed boot 1000ms) *****
Running test suite test_posix_timer
===================================================================
starting test - test_timer
POSIX timer test
Timer fires every 0 secs and  100000000 nsecs
Time remaining to fire 0 secs and  890000000 nsecs
Handler Signal value :20 for 1 times
Handler Signal value :20 for 2 times
Handler Signal value :20 for 3 times
Handler Signal value :20 for 4 times
Handler Signal value :20 for 5 times
Handler Signal value :20 for 6 times
Handler Signal value :20 for 7 times
Handler Signal value :20 for 8 times
Handler Signal value :20 for 9 times
Handler Signal value :20 for 10 times
Handler Signal value :20 for 11 times
Handler Signal value :20 for 12 times

    Assertion failed at tests/posix/timer/src/posix_timer.c:87: test_timer: (total_secs_timer not equal to secs_elapsed)
POSIX timer test has failed
FAIL - test_timer
===================================================================
===================================================================
RunID: ci-180524-0741-1662:knj5
PROJECT EXECUTION FAILED
```

pizi-nordic, 2018-05-30:
Hi.

It looks that the problem is here:
```c
int clock_gettime(clockid_t clock_id, struct timespec *ts)
    s64_t elapsed_msecs, elapsed_secs;
    s64_t elapsed_nsec, elapsed_cycles;

    if (clock_id != CLOCK_MONOTONIC) {
        errno = EINVAL;
        return -1;
    }

    elapsed_msecs = k_uptime_get();
    elapsed_secs = elapsed_msecs / MSEC_PER_SEC;

    elapsed_cycles = (s64_t) (k_cycle_get_32() %
                  sys_clock_hw_cycles_per_sec);
    elapsed_nsec = (s64_t) ((elapsed_cycles * NSEC_PER_SEC) /
                sys_clock_hw_cycles_per_sec);

    ts->tv_sec = (s32_t) elapsed_secs;
    ts->tv_nsec = (s32_t) elapsed_nsec;
    return 0;
}
```

As you see, the output of the function is calculated basing on 2 separate clock sources - the k_uptime_get() and k_cycle_get_32(), which generates several problems.

Please note, that timestamp is not obtained in atomic way. There could be a timer interrupt between `k_uptime_get()` and `k_cycle_get_32()` which might result in dropping one second of time (elapsed cycles will drop to zero but elapsed_secs will still old old value). Disabling interrupts does not help much, as it only freezes time returned by `k_uptime_get()`.

nashif, 2018-05-30:
@ramakrishnapallala fyi

nashif, 2018-06-02:
fixed by #8124

closed 2018-06-02 by nashif