zephyrproject-rtos/zephyr #7906: tests/benchmarks/sys_kernel fails on Arduino Due

inakypg, 2018-05-25:
commit 8e41572774ec72d623d48b1a69e362b00771ff03

Console reports TEST RESULT: FAILED DETAILS: timer tick happened. Results are inaccurate

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.0-rc1 (delayed boot 1000ms) *****

MODULE: kernel API test
KERNEL VERSION: 0x10c0000

Each test below is repeated 5000 times;
average time for one iteration is displayed.

TEST CASE: Semaphore #1
TEST COVERAGE:
	k_sem_init
	k_sem_take(K_FOREVER)
	k_sem_give
Starting test. Please wait...c
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: Semaphore #2
TEST COVERAGE:
	k_sem_init
	k_sem_take(TICKS_NONE)
	k_yield
	k_sem_give
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: Semaphore #3
TEST COVERAGE:
	k_sem_init
	k_sem_take(K_FOREVER)
	k_sem_give
	k_sem_give
	k_sem_take(K_FOREVER)
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: LIFO #1
TEST COVERAGE:
	k_lifo_init
	k_lifo_get(K_FOREVER)
	k_lifo_put
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: LIFO #2
TEST COVERAGE:
	k_lifo_init
	k_lifo_get(K_FOREVER)
	k_lifo_get(TICKS_NONE)
	k_lifo_put
	k_yield
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: LIFO #3
TEST COVERAGE:
	k_lifo_init
	k_lifo_get(K_FOREVER)
	k_lifo_put
	k_lifo_get(K_FOREVER)
	k_lifo_put
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: FIFO #1
TEST COVERAGE:
	k_fifo_init
	k_fifo_get(K_FOREVER)
	k_fifo_put
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: FIFO #2
TEST COVERAGE:
	k_fifo_init
	k_fifo_get(K_FOREVER)
	k_fifo_get(TICKS_NONE)
	k_fifo_put
	k_yield
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: FIFO #3
TEST COVERAGE:
	k_fifo_init
	k_fifo_get(K_FOREVER)
	k_fifo_put
	k_fifo_get(K_FOREVER)
	k_fifo_put
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: Stack #1
TEST COVERAGE:
	k_stack_init
	k_stack_pop(K_FOREVER)
	k_stack_push
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: Stack #2
TEST COVERAGE:
	k_stack_init
	k_stack_pop(K_FOREVER)
	k_stack_pop
	k_stack_push
	k_yield
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: Stack #3
TEST COVERAGE:
	k_stack_init
	k_stack_pop(K_FOREVER)
	k_stack_push
	k_stack_pop(K_FOREVER)
	k_stack_push
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

PROJECT EXECUTION FAILED
RunID: ci-180524-1825-1667:41hi
```

andrewboie, 2018-06-11:
I think we might want to raise the priority of this one, we really need to make sure we're doing the right thing in the ARM timer code

nniranjhana, 2018-06-14:
This failure is also seen on frdm_k64f and sam_e70 platforms.

ManojSubbarao, 2018-06-14:
@AdithyaBaglody , please update the latest status

ManojSubbarao, 2018-08-21:
@AdithyaBaglody Please take a quick look and update/close on this issue. Pending from long time.

closed 2018-08-29 by nashif