zephyrproject-rtos/zephyr #7909: tests/kernel/common.test_bitfield fails on max10

inakypg, 2018-05-25:
commit 8e41572774ec72d623d48b1a69e362b00771ff03

```
***** Booting Zephyr OS v1.11.0-1524-g72a0971dab (delayed boot 1000ms) *****
Running test suite common
===================================================================
starting test - test_byteorder_memcpy_swap
PASS - test_byteorder_memcpy_swap
===================================================================
starting test - test_byteorder_mem_swap
PASS - test_byteorder_mem_swap
===================================================================
starting test - test_atomic
PASS - test_atomic
===================================================================
starting test - test_bitfield
twiddling bits....

    Assertion failed at tests/kernel/common/src/bitfield.c:36: test_bitfield: (b1 not equal to 0)
sys_clear_bit failed for bit 0

FAIL - test_bitfield
===================================================================
starting test - test_printk
1 4294967295 ffffffff22 113 10000 32768 40000 22
p 112 -10000 -32768 -40000 -22
0xcafebabe 0x0000beef
0x1 0x01 0x0001 0x00000001
0x1 0x 1 0x   1 0x       1
42 42 0042 00000042
-42 -42 -042 -0000042
42 42   42       42
42 42 0042 00000042
255     42    abcdef  0x0000002a      42
-1 4294967295 ffffffff
PASS - test_printk
===================================================================
starting test - test_slist
PASS - test_slist
===================================================================
starting test - test_dlist
PASS - test_dlist
===================================================================
starting test - test_intmath
PASS - test_intmath
===================================================================
starting test - test_timeout_order
thread 0 synced on timer 0
thread 1 synced on timer 1
thread 2 synced on timer 2
PASS - test_timeout_order
===================================================================
starting test - test_clock_uptime
PASS - test_clock_uptime
===================================================================
starting test - test_clock_cycle
PASS - test_clock_cycle
===================================================================
starting test - test_version
PASS - test_version
===================================================================
starting test - test_multilib
PASS - test_multilib
===================================================================
===================================================================
RunID: :8mqh
PROJECT EXECUTION FAILED

```

spoorthik, 2018-09-07:
The test case fails on RC3: 

```
***** Booting Zephyr OS v1.13.0-rc3 (delayed boot 3000ms) *****
Running test suite common
===================================================================
starting test - test_byteorder_memcpy_swap
PASS - test_byteorder_memcpy_swap
===================================================================
starting test - test_byteorder_mem_swap
PASS - test_byteorder_mem_swap
===================================================================
starting test - test_atomic
PASS - test_atomic
===================================================================
starting test - test_bitfield
twiddling bits....

    Assertion failed at /home/kspoorth/work/latest_zephyr/tests/kernel/common/src/bitfield.c:43: test_bitfield: (b1 not equal to (1 << bit))
sys_set_bit failed on bit 0

FAIL - test_bitfield
===================================================================
starting test - test_printk
1 4294967295 ffffffff22 113 10000 32768 40000 22
p 112 -10000 -32768 -40000 -22
0xcafebabe 0x0000beef
0x1 0x01 0x0001 0x00000001
0x1 0x 1 0x   1 0x       1
42 42 0042 00000042
-42 -42 -042 -0000042
42 42   42       42
42 42 0042 00000042
255     42    abcdef  0x0000002a      42
-1 4294967295 ffffffff
PASS - test_printk
===================================================================
starting test - test_slist
PASS - test_slist
===================================================================
starting test - test_dlist
PASS - test_dlist
===================================================================
starting test - test_intmath
PASS - test_intmath
===================================================================
starting test - test_timeout_order
thread 0 synced on timer 0
thread 1 synced on timer 1
thread 2 synced on timer 2
PASS - test_timeout_order
===================================================================
starting test - test_clock_uptime
PASS - test_clock_uptime
===================================================================
starting test - test_clock_cycle
PASS - test_clock_cycle
===================================================================
starting test - test_version
PASS - test_version
===================================================================
starting test - test_multilib
PASS - test_multilib
===================================================================
===================================================================

PROJECT EXECUTION FAILED

```

ramakrishnapallala, 2019-02-25:
@inakypg is this TCF issue or does it happen normally?

closed 2019-11-19 by nashif