zephyrproject-rtos/zephyr #7931: Bluetooth controller nrf52 - connection event status

spoerk, 2018-05-25:
I have a question regarding the _Bluetooth controller implementation_ on the Nordic Semiconductor _nrf52 platform_. Is there any way I can evaluate the _status_ of an _individual BLE connection event_ of an established connection?

During a BLE connection there are several connection events where the BLE master and slave are not able to exchange packets (e.g., due to external radio interference). I want to get the state of each individual event. For example, on the Texas Instruments CC2650 there was a status code for every BLE connection event that indicaded that the connection event was OK or that BLE radio did not get a sync during this specific connection event.

carlescufi, 2018-05-28:
Hi @spoerk. I assume that you mean on-air radio connection events. This is not possible today directly as it is not part of the Bluetooth specification. We would require a specific interface between Host and Link Layer to achieve this.
I will let @cvinayak comment on this.

spoerk, 2018-05-28:
Thank you for the response. Yes, I mean on-air connection events.
I am aware that this behavior is not forseen by the Bluetooth specification. It would be fine, if I could evaluate the status of individual connection events in the Bluetooth controller implementation of the nRF52. I am just curious if the nRF52 radio interface does provide such information.

closed 2018-06-04 by spoerk