zephyrproject-rtos/zephyr #7939: Connections to TI CC254x break after conn_update

jonpry, 2018-05-25:
Nordic softdevices have a flag BLE_GAP_OPT_COMPAT_MODE, for ensuring the connection window offset does not become zero because of bugs in TI CC254x stacks. Is there a way to prevent such windows from being used in zephyr?

carlescufi, 2018-05-31:
I've assigned this to @cvinayak so he can provide an answer here.

jonpry, 2018-05-31:
I put the following in isr_rx_scan()

		if (_radio.scanner.win_offset_us == 0) {
		   _radio.scanner.win_offset_us = 2;
		}		
                {
			conn_space_us = _radio.scanner.win_offset_us;
			while ((conn_space_us & ((u32_t)1 << 31)) ||
			       (conn_space_us < conn_offset_us)) {
				conn_space_us += conn_interval_us;
			}
			pdu_adv_tx->connect_ind.win_offset =
				(conn_space_us - conn_offset_us) / 1250;
			pdu_adv_tx->connect_ind.win_size++;
		}

Now connections to TI devices work, but i doubt this is safe to do. I also considered calling the ticker api and reserving the first time slot for some non existent user function, but wasn't able to figure out the ticker api or the correct constants to make that work.

closed 2020-10-06 by github-actions[bot]