zephyrproject-rtos/zephyr #7986: The scripts (debug, debugserver and flash) are not working for Intel S1000 board

rgundi, 2018-05-29:
This was working before. Looks like the commit bc7b1c595ed4090be6d80859a01a0bb372ae030c regressed it.

mbolivar, 2018-05-31:
@rgundi the surface cause of the issue is that the flasher has internally moved the "common" configuration values like the board directory, kernel elf, etc. away from what you can set with board_finalize_runner_args(). I think this is **not** the root cause of the issue.

I took a look at boards/common/intel_s1000_crb.board.cmake (by the way, this entire file should just be inlined into boards/xtensa/intel_s1000_crb/board.cmake, since there is nothing "common" about its contents).

I see some suspicious lines that appear to be the real root cause, rather than the west changes:

https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/common/intel_s1000_crb.board.cmake

Specifically:

- setting `--board-dir` would have no effect even if it worked, since it's being set to the default value
- ditto for `--kernel-elf`, and that's hard-coding something that is already configured elsewhere by the build system
- as far as I can tell, ditto for `--gdb`, the file appears to be taking the existing setting already set in gcc.cmake (since xcc.cmake sets COMPILER to gcc)

Since I can't build for your board as it does not have a publicly available toolchain, I am just guessing here.

I have submitted a PR that I think fixes these issues (#8076); could you please test?

Thanks!

rgundi, 2018-06-01:
@mbolivar .. As you have already figured out by now, I have fixed the runner related changes in #7988.  However, i can do debugserver, debug and flash successfully only if i revert the commit bc7b1c5.

mbolivar, 2018-06-01:
@rgundi I see. Thanks for your work locating this commit as a problem.

Do you have some more details, since I can't run this without the board and access to the toolchain?

Output from these commands (when run from your build directory) would be useful:

```
west -vvv flash
```

```
west -vvv debug
```

```
west -vvv debugserver
```

nashif, 2018-06-05:
@rgundi can you please provide the requested information? @sathishkuttan can you help with this as well?

nashif, 2019-02-10:
this actually works, but I would not call this flashing, the script needs to be adapted to do flashing over USB which this is supported. Closing this for now.

closed 2019-02-10 by nashif