zephyrproject-rtos/zephyr #7989: eth: ping: can't ping others from zephyr console

zhaozhongchen, 2018-05-29:
my custom board(ip: 192.168.3.25) can't ping others(ip: 192.168.3.101, win10) in console.

```
net> ping 192.168.3.101                                                               
Sent a ping to 192.168.3.101                                                          
Ping timeout                                                                          
net> 
```

but from my win10 computer can ping Zephyr:

```
C:\Users\qianfan>ping 192.168.3.25  -t

正在 Ping 192.168.3.25 具有 32 字节的数据:
来自 192.168.3.25 的回复: 字节=32 时间=1ms TTL=128
来自 192.168.3.25 的回复: 字节=32 时间=1ms TTL=128
来自 192.168.3.25 的回复: 字节=32 时间=1ms TTL=128
来自 192.168.3.25 的回复: 字节=32 时间=1ms TTL=128
来自 192.168.3.25 的回复: 字节=32 时间=1ms TTL=128
来自 192.168.3.25 的回复: 字节=32 时间=1ms TTL=128

192.168.3.25 的 Ping 统计信息:
    数据包: 已发送 = 6，已接收 = 6，丢失 = 0 (0% 丢失)，
往返行程的估计时间(以毫秒为单位):
    最短 = 1ms，最长 = 1ms，平均 = 1ms
Control-C
^C
```

zhaozhongchen, 2018-05-29:
ping win10 from zephyr's console continued, seems the first time after ARP failed and the next time successful.

```
shell> select net                                                                                
net> ping 192.168.3.101                                                                          
Sent a ping to 192.168.3.101                                                                     
Ping timeout                                                                                     
net> ping 192.168.3.101                                                                          
Sent a ping to 192.168.3.101                                                                     
Received echo reply from 192.168.3.101 to 192.168.3.25                                           
net> ping 192.168.3.101                                                                          
Sent a ping to 192.168.3.101                                                                     
Ping timeout                                                                                     
net> ping 192.168.3.101                                                                          
Sent a ping to 192.168.3.101                                                                     
Received echo reply from 192.168.3.101 to 192.168.3.25                                           
net> ping 192.168.3.101                                                                          
Sent a ping to 192.168.3.101                                                                     
Received echo reply from 192.168.3.101 to 192.168.3.25                                           
net> ping 192.168.3.101                                                                          
Sent a ping to 192.168.3.101                                                                     
Ping timeout                                                                                     
net> 
```

tbursztyka, 2018-05-30:
According to your wireshark logs, it's the responses that are not sent, so the issue does not seem to be on your Zephyr enabled board.

zhaozhongchen, 2018-05-30:
@tbursztyka yes zephyr has sent ICMP packet but doens't get reply sometimes.

tbursztyka, 2018-05-30:
yes, and according to your wireshark logs the response is not sent by the destination (win10), so either your win10 does not receives it or something else, but zephyr does not seem to have any issue here.

zhaozhongchen, 2018-05-30:
but sometimes ping successful and sometimes failed.

zhaozhongchen, 2018-05-31:
@tbursztyka @jukkar I had enable "Validata the IPv4 checksum if possible" option and found the checksum in IP part is illeage. please refer packet **47** 

```
Frame 47: 60 bytes on wire (480 bits), 60 bytes captured (480 bits) on interface 0
Ethernet II, Src: Stmicroe_69:00:be (00:80:e1:69:00:be), Dst: IntelCor_d6:eb:a1 (e0:94:67:d6:eb:a1)
Internet Protocol Version 4, Src: 192.168.3.25, Dst: 192.168.3.101
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
    Total Length: 28
    Identification: 0x0000 (0)
    Flags: 0x00
    Fragment offset: 0
    Time to live: 64
    Protocol: ICMP (1)
    Header checksum: 0xd5cd incorrect, should be 0xf312(may be caused by "IP checksum offload"?)
        [Expert Info (Error/Checksum): Bad checksum [should be 0xf312]]
    [Header checksum status: Bad]
    [Calculated Checksum: 0xf312]
    Source: 192.168.3.25
    Destination: 192.168.3.101
    [Source GeoIP: Unknown]
    [Destination GeoIP: Unknown]
Internet Control Message Protocol
```

zhaozhongchen, 2018-05-31:
I print all IPv4's data in calc_chksum in this way:

```
--- a/subsys/net/ip/utils.c
+++ b/subsys/net/ip/utils.c
@@ -379,6 +379,14 @@ static u16_t calc_chksum(u16_t sum, const u8_t *ptr, u16_t len)
        const u8_t *end;
 
        end = ptr + len - 1;
+       const u8_t *p = ptr;
+
+       if (len == 20) {
+               while (p <= end) {
+                       printk("%02x ", *p++);
+               }
+               printk("\n");
+       }
```

If I always ping a const ip address, the data in IPv4 part should be fixed, but it doesn't. 

```
shell> select net
net> ping 192.168.3.101
45 00 00 1c 00 00 00 00 40 01 00 00 00 00 00 00 00 00 00 00
Sent a ping to 192.168.3.101
Ping timeout
net> ping 192.168.3.101
45 00 00 1c 00 00 00 00 40 01 00 00 c0 a8 03 19 c0 a8 03 65
Sent a ping to 192.168.3.101
Received echo reply from 192.168.3.101 to 192.168.3.25
net> ping 192.168.3.101
45 00 00 1c 00 00 00 00 40 01 00 00 01 d2 c0 a8 03 19 e0 94
Sent a ping to 192.168.3.101
Ping timeout
```

AND I had fixed this issue in #8058

tbursztyka, 2018-05-31:
That clarifies indeed, good catch

PR #8058 (2018-05-31): net: icmpv4: fix incorrect IP header checksum
IPv4 header's checksum was set when 'setup_ipv4_header', but in that
times IPv4 header's source ip and dest ip address haven't setted. So
it may be set an incorrect checksum.

Fixes: #7989
commit c84b37b0 (2018-05-31): net: icmpv4: fix incorrect IP header checksum

IPv4 header's checksum was set when 'setup_ipv4_header', but in that
times IPv4 header's source ip and dest ip address haven't setted. So
it may be set an incorrect checksum.

Fixes: #7989
closed 2018-05-31 by nashif