zephyrproject-rtos/zephyr #8013: Open-AMP：power on can not communicate

Jeepgoing, 2018-05-30:
this issus appeared when i test sample/subsys/ipc/openamp demo.
first step: download image to flash through make flash.
second step: press reset switch on LPCXpresso5411X board and the following message appeared on serial port, as below:

  ***** Booting Zephyr OS v1.12.0-rc1-41-gfcc56e3 *****
  Starting application thread!

  OpenAMP demo started
   Primary core received a message: 1
   Primary core received a message: 3
   Primary core received a message: 5
   ...
   Primary core received a message: 101
   OpenAMP demo ended.

third step: disconnect the power supply of the development board, and then power on board, i found m4 can not communicate with m0, serial port message as below:

  ***** Booting Zephyr OS v1.12.0-rc1-41-gfcc56e3 *****
  Starting application thread!

OpenAMP demo started

so why m4 can not communicate m0 when i power on the board?

Jeepgoing, 2018-05-31:
hi all,
i am now using nxp lpc54102, have we support this chip in the future? lpc54114's flash is 256k, i need mcuboot+parameter storage+m4 image+m0 image, i think it is difficult to OTA. lpc54102 have 512k flash, i can design application program more easily with it.

galak, 2018-05-31:
> i am now using nxp lpc54102, have we support this chip in the future? lpc54114's flash is 256k, i need mcuboot+parameter storage+m4 image+m0 image, i think it is difficult to OTA. lpc54102 have 512k flash, i can design application program more easily with it.

You can open an issue asking for support for the lpc54102.  I'd assume it would be too difficult based on the support we have for LPC54114.  @MaureenHelm can comment further.

stanislav-poboril, 2018-05-31:
I have tried the RPMsg-Lite demo (which is not in the repository). Modified it to send a million of messages. When connected USB cable, which powers the board, and opened console afterwards, I could see the stream of messages.

When I did the similar thing with Open-AMP demo, it did not output anything (the initial banner etc. could be already sent before, which is expected, but there was no message output, which is not ok). It worked with the reset button.

When I added some k_sleep() call into app_task (either remote or master), it did not work even after reset button. Note that the second core image is loaded before main(), so the timing could affect initialization of communication.

Even with RPMsg-Lite demo, if I upload using on-board debugger and JLink SW, I see only the initial banner and message and have to press reset button (or power off/on completely) to see the subsequent output.

MaureenHelm, 2018-06-01:
It should be straightforward to port to lpc54102 using lpc54114 as an example, and reuse the existing peripheral drivers. To get mcuboot working, a flash driver is also needed. The write-block-size for this flash is significantly larger than other devices (256 bytes vs. 8 bytes on k64), so I'm not sure of the implications to mcuboot.

Pull requests are welcome.

Jeepgoing, 2018-06-02:
i have a question, lpc54114 based on arch/lpc54xxx, but all lpc54102's peripheral drivers are different from lpc54114, so i have to add lpc54102 independence, for example: 

  @@`arch/arm/soc/nxp_lpc/lpc54xxx/soc_lpc54102.c/h
ext/hal/nxp/mcux/drivers/lpc54102
drivers/serial/Kconfig.mcux_lpc54102usart
......`

but i don't think it's a good idea, i usually think lpc54xxx should include both lpc54102 and lpc54114, lpc also should include both lpc54102 and lpc54114. do you have any good suggestions about how to add lpc54102? @MaureenHelm

Jeepgoing, 2018-06-07:
i have open an new issue about how to add lpc54102 board.

closed 2018-06-12 by Jeepgoing