zephyrproject-rtos/zephyr #8015: Driver:spi_flash_w25qxxdv.c init mistake and can not be erased

simonccn, 2018-05-30:
1. one mistake code in spi_flash_init()
`ret = spi_flash_wb_configure(dev);
if (ret) {
	dev->driver_api = &spi_flash_api;
}`

 here ret should be 0 when spi_flash_wb_configure() return success, so the code change to:
`ret = spi_flash_wb_configure(dev);
if (ret == 0) {
	dev->driver_api = &spi_flash_api;
}`

2. spi_flash_wb_erase_internal() invokes spi_flash_wb_access() with parameter error.
`return spi_flash_wb_access(driver_data, erase_opcode,true, offset, NULL, len, true);`

here len must be 0, otherwise flash can't be erased correctly.

SavinayDharmappa, 2018-05-30:
> ret = spi_flash_wb_configure(dev); if (ret) { dev->driver_api = &spi_flash_api; }

It is fixed a6d8378377dfba7392f7632bc7acc3438fcefdb4

tbursztyka, 2018-05-30:
that's fixed.

closed 2018-05-30 by tbursztyka