zephyrproject-rtos/zephyr #8018: ARMv8-M: MPU driver implementation

ioannisg, 2018-05-30:
ARM MPU for ARMv8-M

1. There are architectural changes in ARMv8-M.
    - Now, the size of a region can be fully flexible, i.e. does not need to be a power of 2.
    - A region can start from any address (32-byte aligned) and end at any address.
    - As a result, the regions are, now, not allowed to overlap

2. There are register API changes
   - Some registers are the same
      i. TYPE
     ii. CTRL
    iii. RNR
   - Some registers are different
     i. RBAR: Contains the Base Address (present in ARMv7), does not contain REGION and VALID (present in ARMv7), and contains attributes (not present in ARMv7)
   - Some registers are new
     i. RLAR
   - Some register do not exist
     i. RASR

3. The possible permission-configuration options are limited in ARMv8
   a. the Privileged-RW/Unprivileged-RO does not exit
   b. the No-Access does not exist

---

We need to adapt the ARM_MPU driver (HAL access), so it works both for ARMv7-M and ARMv8-M architectures.
We need to refactor the ARM MPU internal API (e.g. macros for region/attribute definitions), so it can work with both architectures.

ioannisg, 2018-05-31:
@carlescufi @agross-linaro

ioannisg, 2018-06-26:
Some more detailed comments regarding the architectural changes we need:

-  In ARMv8-M, MPU regions _cannot_ overlap. This means that we cannot define, configure, and enable the _background_ regions upon MPU initialization during system boot.
   - We can still define _partition_ regions during system boot, provided that the regions will not overlap with other MPU regions that may be configured & enabled at a later point.
   - The above could be the case for the region definition for the Flash partition.

andrewboie, 2018-06-26:
We need to consider the execute bit as well. Current policy has XIP systems in mind.

Currently, at least with the arm_mpu, the "background" map is RWX for supervisor and no access to user.
We configure SRAM_* to be RW for supervisor and no access for user, execute bit disabled.
We configure FLASH_* to be RX for both supervisor and user, except if the config flash write thing is turned on in which case supervisor get RWX to flash.

Since we can't have overlapping regions in SRAM, is there a way to disable execution in the background map for ARMv8? And then just enable execution for the flash region?

ioannisg, 2018-07-02:
@andrewboie, thanks for your feedback

>Since we can't have overlapping regions in SRAM, is there a way to disable execution in the background map for ARMv8? And then just enable execution for the flash region?

As far as I know the only way to do these, is to define explicit MPU regions. But then, for SRAM this region would overlap with dynamically configured regions, e.g. for User-space, Stack-Guards etc.

For Flash, we _can_ define RX for both user and supervisor, I believe, since we do not expect to dynamically configure any regions.

ioannisg, 2018-07-06:
Pull-request for the driver implementation: #8785.

ioannisg, 2018-08-14:
Pull-request for migration into the new logging system: #9361

ioannisg, 2018-10-03:
The non-overlapping MPU regions' issue is handler in #8907

ioannisg, 2018-11-13:
Pull-request for non-overlapping MPU regions: #11346

closed 2019-02-05 by ioannisg