zephyrproject-rtos/zephyr #8033: tests/crypto/mbedtls/ results in exception on frdm_k64f

pswarnak, 2018-05-30:
The test **tests/crypto/mbedtls/** results in exception.

**Arch:** arm
**Board:** frdm_k64f
**Zephyr tag:**  v1.12.0-rc2 - commit d93ecda4c590d674e1cccdc65fb3501ae80a2055.

**Error Console Log:**
```
***** Booting Zephyr OS v1.12.0-rc2 *****
***** BUS FAULT *****
  Executing thread ID (thread): 0x20004138
  Faulting instruction address:  0xe1f264a0
  Stacking error
Fatal fault in essential thread! Spinning...
```

**Steps to reproduce:**
```
cd tests/crypto/mbedtls/
rm -rf build && mkdir build && cd build
cmake -D BOARD=frdm_k64f ../
make BOARD=frdm_k64f
cp zephyr/zephyr.bin /run/media/pswarnak/MBED/

Check the console log.
```

MaureenHelm, 2018-05-30:
Bisected to ba2c44e4839af520008cbb5ee8a31983e2f967b8

```
Author: Andrew Boie <andrew.p.boie@intel.com>
Date:   Thu May 24 22:14:20 2018 -0700

    tests: enable HW stack protection by default
    
    Userspace doesn't necessarily imply stack overflow protection
    for supervisor threads. Enable this as well if the hardware
    supports it.
    
    Signed-off-by: Andrew Boie <andrew.p.boie@intel.com>
```

andrewboie, 2018-05-30:
Is this test overflowing its stack? Seems like most likely explanation.

closed 2018-05-30 by MaureenHelm

pswarnak, 2018-05-31:
Thanks @andrewboie. Validated the test, works fine now.

PR #8037 (2018-05-30): tests: mbedtls: don't use stdout console
This redirects PRINT() to printf() and not printk() and
was causing stack overflows on some platforms.

Fixes: #8033
commit aa26d992 (2018-05-30): tests: mbedtls: don't use stdout console

This redirects PRINT() to printf() and not printk() and
was causing stack overflows on some platforms.

Fixes: #8033