zephyrproject-rtos/zephyr #8039: tests/shell failing on Arduino 101 / Quark SE arc

inakypg, 2018-05-30:
Fails to boot and produce any console output; same result on 11 different boards (6 Quark SE, 5 A101) (not even the " booting Zephyr message")

commit v1.12.0-rc2-7-g55494c3f0a

vonhust, 2018-05-31:
@inakypg  Could you list the error log?

inakypg, 2018-06-01:
@vonhust it doesn't produce any output, hence why I didn't list it.

lpereira, 2018-06-02:
I'm not able to reproduce this issue.  Just built tests/shell for arduino_101, flashed it, and got the expected output:

```
***** Booting Zephyr OS v1.12.0-rc2-68-g158ea970ea *****
Zephyr Shell, Zephyr version: 1.12.0-rc2
Type 'help' for a list of available commands
shell> Running test suite shell_test_suite
===================================================================
starting test - test_help
[Modules]
dummy

[Commands]
noprompt

To select a module, enter 'select <module name>'.
shell_exec(help): 0
help
cmd1
cmd2                         <arg1>
cmd3                         <arg1> <arg2>

Enter 'exit' to leave current module.
shell_exec(help dummy): 0
No help found for 'invalid'
Usage: help [command]
shell_exec(help invalid): -22
PASS - test_help
(...)
```

lpereira, 2018-06-02:
Ah, this is on x86.  Let me try the arc port.

tbursztyka, 2018-06-04:
@inakypg can you retry? It seems to be fixed. I guess commit ID 05063ede55  fixed it.

nashif, 2018-07-20:
fixed..

closed 2018-07-20 by nashif