zephyrproject-rtos/zephyr #8054: Ethernet initialization is unreliable and gets stuck on frdm-k64f

vakulgarg, 2018-05-31:
I am using ethernet on frdm-k64f. I find that a lot of times, the ethernet phy initialization function eth_mcux_phy_start() gets stuck (i.e. never returns). This prevents zephyr boot (the shell prompt never appears).

I tried adding a printf before eth_mcux_phy_start() is called in function eth_0_init(). This effectively inserts some delay. Doing this workaround the problem. Alternatively, adding k_busy_wait(1000000) also workaround the issue.

mike-scott, 2018-05-31:
Which sample are you running?

Could it be that you're experiencing this:
https://github.com/zephyrproject-rtos/zephyr/issues/8049 ?
(more detail in https://github.com/zephyrproject-rtos/zephyr/issues/8005)

It manifests as a hung boot right after the ethernet DBG message.

vakulgarg, 2018-05-31:
I am running mbedtls_sslclient (prj.conf modified to use ethernet).
It is hung boot right after ethernet DBG message.

vakulgarg, 2018-05-31:
Your patch below worked for me.
https://github.com/mike-scott/zephyr/commit/4054d85d44fe48f6594a7dfeea6cf7f5af1b761f

mike-scott, 2018-05-31:
@nashif Can we mark this as a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/8049 ?

MaureenHelm, 2018-05-31:
Duplicate of #8049

mike-scott, 2018-05-31:
https://github.com/zephyrproject-rtos/zephyr/pull/8077 is merged and should fix this as well (as it's a duplicate).

Can @vakulgarg confirm that latest master does not exhibit hung boot?

vakulgarg, 2018-06-01:
With latest master, it works fine.

closed 2018-06-01 by vakulgarg

pfalcon, 2018-06-04:
> I find that a lot of times, the ethernet phy initialization function eth_mcux_phy_start() gets stuck (i.e. never returns).

Funnily, I never saw this with my samples (which are usually from samples/net/socksts/). Just trying to compare my experiences with other people's (given that everyone seems to see different picture).