zephyrproject-rtos/zephyr #8065: tests/subsys/fs/fat_fs_api - test_fat_file and test_fat_dir results into Assertion failure on Arduino_101 due to spi_flash changes

pswarnak, 2018-05-31:
The test **tests/subsys/fs/fat_fs_api** results in Assertion failure in test_fat_file and test_fat_dir sub-test cases. 

**Arch:** x86
**Board:** arduino_101
**Zephyr tag:**  v1.12.0-rc2 - commit d93ecda4c590d674e1cccdc65fb3501ae80a2055.

**Error Console Log:**
```
***** Booting Zephyr OS v1.12.0-rc2-19-g386b3e6 *****
Running test suite fat_fs_basic_test
===================================================================
starting test - test_fat_mount
PASS - test_fat_mount
===================================================================
starting test - test_fat_file

Open tests:
Creating new file /NAND:/testfile.txt
Opened file /NAND:/testfile.txt

Write tests:
Data written:"hello world!"

Data successfully written!

Sync tests:

Read tests:
Data read:"hello world!"

Data read matches data written

Truncate tests:

Testing shrink to 0 size
Testing write after truncating

Write tests:
Data written:"hello world!"

Data successfully written!
Original size of file = 12                                                                                            
                                                                                                                      
Testing shrinking                                                                                                     
fs_truncate failed [-5]                                                                                               
                                                                                                                  
    Assertion failed at ../src/test_fat_file.c:279: test_fat_file: test_file_truncate() == TC_PASS is false

FAIL - test_fat_file
===================================================================
starting test - test_fat_dir

mkdir tests:
Creating new dir /NAND:/testdir

Write tests:
Data written:"hello world!"

Data successfully written!
Created dir /NAND:/testdir!

lsdir tests:

Listing dir /NAND::
[DIR ] BT
[FILE] DESTDI@ (size = 0)

lsdir tests:
Error opening dir /NAND:/testdir [-2]

    Assertion failed at ../src/test_fat_dir.c:163: test_fat_dir: test_lsdir(TEST_DIR) == TC_PASS is false

FAIL - test_fat_dir
===================================================================
starting test - test_fat_fs

Optimal transfer block size   = 512
Allocation unit size          = 1024
Volume size in f_frsize units = 2028
Free space in f_frsize units  = 2027
PASS - test_fat_fs
===================================================================
===================================================================
PROJECT EXECUTION FAILED
```

**Steps to reproduce:**
```
cd zephyr/tests/subsys/fs/fat_fs_api/
rm -rf build && mkdir build && cd build
cmake -D BOARD=arduino_101 ../
make BOARD=arduino_101 flash

Check the console log.
```

**Additional Information:**

test_fat_dir fails in test_lsdir function with /NAND:/testdir
> Error opening dir /NAND:/testdir [-2]

test_fat_file fails in test_file_truncate function on fs_truncate call.
> fs_truncate failed [-5]

pswarnak, 2018-05-31:
CC @ramakrishnapallala @tbursztyka 
Tried further debugging, the test passes by just picking changes till @ramakrishnapallala  commit - c239ea701fe6be5160e90df89283f598c00d6a64.
It fails with the recent changes in spi_flash driver.  This should possibly an issue in SPI flash driver. Kindly check.

ramakrishnapallala, 2018-05-31:
@tbursztyka basically we removed all the spi related changes and kept only the FS related changes. This recipe works fine on arduino_101.

we also ran the same test with altera_max10 which does not have any SPI driver dependencies and it works on that too. (for altera_max10, we have go back to little older code as we see flashing issue with latest code: https://github.com/zephyrproject-rtos/zephyr/issues/7954)

tbursztyka, 2018-05-31:
Let me try this fs test.