zephyrproject-rtos/zephyr #8070: table broken in S1000 documentation

nashif, 2018-05-31:
thats what I see here: http://docs.zephyrproject.org/boards/xtensa/intel_s1000_crb/doc/intel_s1000.html

```
The corresponding pin mapping is +————-+———–+ | Flyswatter2 | S1000 | +=============+===========+ | 1 | 7 | +————-+———–+ | 2 | NC | +————-+———–+ | 3 | 4 | +————-+———–+ | 4 | NC | +————-+———–+ | 5 | 3 | +————-+———–+ | 6 | 8 | +————-+———–+ | 7 | 2 | +————-+———–+ | 8 | NC | +————-+———–+ | 9 | 1 | +————-+———–+ | 10 | NC | +————-+———–+ | 11 | NC | +————-+———–+ | 12 | NC | +————-+———–+ | 13 | 5 | +————-+———–+ | 14 | NC | +————-+———–+ | 15 | NC | +————-+———–+ | 16 | NC | +————-+———–+ | 17 | NC | +————-+———–+ | 18 | NC | +————-+———–+ | 19 | NC | +————-+———–+ | 20 | NC | +————-+———–+

```

rgundi, 2018-05-31:
@nashif is there a way for me to verify the fix before it gets merged? I am not sure what the fix is but i have a couple of ideas.

nashif, 2018-05-31:
make htmldocs? read the documentation please.

PR #8095 (2018-06-01): doc: fix broken table display
Needed a blank like before the table

Fixes: #8070
commit d07d5e87 (2018-06-01): doc: fix broken table display

Needed a blank like before the table

Fixes: #8070
closed 2018-06-01 by nashif