zephyrproject-rtos/zephyr #8081: Bluetooth: Deprecated TinyCrypt and use mbedTLS

lpereira, 2018-05-31:
While TinyCrypt has its merits -- as its name implies, its size -- it's not resilient to some kinds of side-channel attacks.  The fact that it hasn't seen a lot of maintenance recently is also worrying.

I did a quick investigation, and it seems that, at least for Bluetooth -- the only direct use of TinyCrypt APIs that's part of the tree -- it's possible to switch to using mbedTLS instead.  mbedTLS has a drawback, though: it's too big for some devices. So I'm thinking about writing a simple TinyCrypt shim on top of mbedTLS for the devices that can handle it -- and use TinyCrypt for those that don't, until we have a better solution.

@jhedberg, would you be OK with this?

lpereira, 2018-05-31:
Took some time to [write the shim](https://gist.github.com/lpereira/64281f73f91bd4ffba09ae959e30862f); didn't compile or test it yet.  Just needs some Kconfig/CMake magic and this should be all we need to use Bluetooth with mbedTLS.

lpereira, 2018-06-04:
OK, I spent some time on this today.  Fixed some bugs in the shim, and hacked it so that Zephyr is built with mbedTLS rather than TinyCrypt.  Quoting the [commit message](https://github.com/lpereira/zephyr/commit/962ed1c4260cc87d8e3bcb465c2642ab251f7343):

```
    This is a crude hack to use mbedTLS APIs instead of TinyCrypt.  For
    reference, building the mesh example for the bbc_microbit, this is the
    memory usage:

    With mbedTLS:
            Memory region         Used Size  Region Size  %age Used
                       FLASH:      123264 B       256 KB     47.02%
                        SRAM:       15172 B        16 KB     92.60%
                    IDT_LIST:         132 B         2 KB      6.45%

    With TinyCrypt:
            Memory region         Used Size  Region Size  %age Used
                       FLASH:       97368 B       256 KB     37.14%
                        SRAM:       15164 B        16 KB     92.55%
                    IDT_LIST:         132 B         2 KB      6.45%

    So, while there was a ~10% increase in flash usage (text/rodata), this
    seems like a doable approach, at least for this particular test.

    Also, keep in mind that this changes the default mbedTLS configuration
    that's been provided as a minimum to support TLSv1.2, with tweaks to
    add whatever the Bluetooth stack needed; the size can go down if the
    configuration file is configured properly (e.g.  by providing a
    config-bluetooth.h that's specifically tuned to mimic whatever
    TinyCrypt offers for the Bluetooth subsystem).

    Another thing to note is that all shim functions are provided as static
    inline functions in the shim header; flash usage should decrease
    slightly if this is done properly.  This is just a hack to experiment
    with the idea.
```

mike-scott, 2018-06-05:
@lpereira how would this work if we also needed MbedTLS for a different purpose along side the bluetooth usecase?

lpereira, 2018-06-05:
I don't know yet.  Maybe a `config-zephyr.h` for mbedTLS that's the default, and with some ifdefs for features required by other stuff that Zephyr needs? Today the Bluetooth enables certain TinyCrypt features, we could define this in terms of `CONFIG_CRYPTO_NEED_AES`, or `CONFIG_CRYPTO_NEED_ECDH`... I don't know, let me think more about this.

In any case, I think that, for the combined use case of Bluetooth+something else, it's going to be a net win in size.

vakulgarg, 2018-06-05:
Instead of calling mbedtls/tinycrypt apis, shouldn't bluetooth stack call zephyr's crypto driver apis? Then depending upon the  configured shim (tinycrypt or mbedtls), the  required lib would get called.

lpereira, 2018-06-05:
@vakulgarg Yes, that's the idea.  This was just a crude hack to get started.  See the comment with the results.

dleach02, 2018-06-08:
@lpereira, I'm interested in this as well as I'm currently working on wrapping the NXP KW41Z BLE controller library with a shim to get it to work within Zephyr. There are a number of crypto functions I'll need to tie in.

lpereira, 2018-06-08:
If you check out my `bluetooth-mbedtls` branch now, you'll see that the crypto subsystem has been changed.  I've added all the primitives used by the Bluetooth subsystem there, and implemented it for both TinyCrypt and mbedTLS.  I've also began changing the Bluetooth code to use the crypto subsystem instead of making direct calls to TinyCrypt.  There's still a little bit of work to do (I still have to comb the shim implementations to find subtle implementation details), especially regarding testing and Kconfig integration, but I'd like to get this out in the open for early feedback before I open a PR.

nashif, 2019-05-01:
This is not going to happen, tinycrypt is there to stay.

closed 2019-05-01 by nashif