zephyrproject-rtos/zephyr #8083: Bluetooth ATT trying to access invalid pointer after disconnect

danielwidmann, 2018-05-31:
Think of the following scenario on a BLE central device:

* CONFIG_BT_ATT_TX_MAX + 1 GATT requests are issued within a short time (The last request is blocked until one of the other requests complete)
* The BLE connection is lost. `bt_att_disconnected` is called

When a BLE connection is terminated, the ATT layer first wakes up all waiting threads, but they might not be scheduled right away:
https://github.com/zephyrproject-rtos/zephyr/blob/43553da9b2bb9e537185695899fb5184c3b17ebe/subsys/bluetooth/host/att.c#L1967-L1970

Then the used ATT data-structure is cleared:
https://github.com/zephyrproject-rtos/zephyr/blob/43553da9b2bb9e537185695899fb5184c3b17ebe/subsys/bluetooth/host/att.c#L2044

And finally the waiting thread wakes up and tries to release a semaphore that has just been wiped:
https://github.com/zephyrproject-rtos/zephyr/blob/43553da9b2bb9e537185695899fb5184c3b17ebe/subsys/bluetooth/host/att.c#L268-L273

This causes access to invalid memory and a FAULT on the thread.

nashif, 2018-06-05:
@jhedberg FYI, @Vudentz what is the status?

Vudentz, 2018-06-06:
@nashif Will send a fix for this in a moment.

PR #8201 (2018-06-06): Bluetooth: ATT: Fix clearing context at disconnect
When att_disconnected is called a thread may be waiting for the tx_sem
but that is memset to 0, furthermore there exists a flag
ATT_DISCONNECTED to indicate the context is no longer valid so the use
of memset is not really required.

Fixes #8083
commit 1218648e (2018-06-06): Bluetooth: ATT: Fix clearing context at disconnect

When att_disconnected is called a thread may be waiting for the tx_sem
but that is memset to 0, furthermore there exists a flag
ATT_DISCONNECTED to indicate the context is no longer valid so instead
move memset to bt_att_accept so it is cleared when it is about to be
reused.

Fixes #8083
closed 2018-06-06 by nashif