zephyrproject-rtos/zephyr #8097: tests/drivers/watchdog/wdt_basic_api fails on Quark SE / x86

inakypg, 2018-06-01:
Toolchain: ZEPHYR_GCC_VARIANT=zephyr

Commit 43553da9b2bb9e537185695899fb5184c3b17ebe

Testcase seems to never complete after restarting the system

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.0-rc2 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.0-rc2 (delayed boot 1000ms) *****
Running test suite wdt_basic_test
===================================================================
starting test - test_wdt
Testcase: test_wdt_no_callback
```

tbursztyka, 2018-06-04:
I don't know why this was assigned to me, but running a git bisect (I put the good point on 1.11) points to commit 7a5ff13703a0139f917afd58151934b5fcf19bb9

@nashif Can you check?

That said, while running the git bisect, I realized that test (which is made only for qmsi... that sucks, moreover it's supposed to test the generic API but anyway) gives a slight different output on success sometimes: sometime it resets and pass, sometime it does not reset but still passes.

nashif, 2018-06-04:
r

> @nashif Can you check?

i just hit this commit with bisect myself, looking

nashif, 2018-07-24:
@punitvara  status update please? any fix in sight?

ManojSubbarao, 2018-08-21:
@punitvara Please take a quick look and drive this issue to closure. This is pending from long time without any updates.

SavinayDharmappa, 2018-08-28:
@nashif  I did some intial analysis, the watch dog timer for qmsi and esp32 are not moved to new interfaces introduced . But application uses newly introduced interface for which the QMSI and esp32 WDT driver are not compaitable. This reason why esp32 crashes. Still wondering why null pointer derefercing of QMSI driver is not showing crash log instead just hangs.

markusr, 2018-08-31:
I also have a problem with watchdog example on STM32 (```nucleo_l432kc```). 

```
***** Booting Zephyr OS v1.13.0-rc2-1-g19b7eaa *****
***** USAGE FAULT *****
  Illegal use of the EPSR
***** Hardware exception *****
Current thread ID = 0x2000022c
Faulting instruction address = 0x80200288
Fatal fault in essential thread! Spinning...
***** Booting Zephyr OS v1.13.0-rc2-1-g19b7eaa *****
***** USAGE FAULT *****
  Illegal use of the EPSR
***** Hardware exception *****
Current thread ID = 0x2000022c
Faulting instruction address = 0x80200288
Fatal fault in essential thread! Spinning...
```

SavinayDharmappa, 2018-09-03:
@nashif @ManojSubbarao  Modified wdt driver to new interfaces defined in watchdog.h. As of now only no callback test case is passing other two test cases are failing. I am still debugging the issue.

SavinayDharmappa, 2018-09-05:
After wdt qmsi driver changes , test cases for watchdog are passing. I have tested it on quark_se_c1000 board. I will work on esp32 driver changes as this driver also needs to be migrated for new wdt API interfaces.

SavinayDharmappa, 2018-09-06:
Implemented new wdt api interface for esp32. After that test case are failing, because global data are re-initialized upon system reset due watchdog time expiry. I am still debugging. This not the case for quark_se_c100.

SavinayDharmappa, 2018-09-10:
https://github.com/zephyrproject-rtos/zephyr/pull/9872

ManojSubbarao, 2018-10-22:
@SavinayDharmappa  please look into this issue

SavinayDharmappa, 2019-02-11:
This issue is reproduced only in quark d2000 and arc of quark_se_c1000 . The driver changes for x86 quark_se_c1000 part  still debugging to find the root cause.

SavinayDharmappa, 2019-02-21:
In case of quark_d2000 watchdog reset is triggered , but variable defined in the app_noinit section also getting reinitialized during watchdog reset, which not the case for other platforms. Because of which "test_wdt_no_callback" repeats indefinitely.

SavinayDharmappa, 2019-02-21:
After further debugging it clear from status register reset is triggered by watchdog only but ram contents are being lost in that reset. Because of which test is going repeated loop. I think we have bypass this testcase for quark_d2000 @nashif

SavinayDharmappa, 2019-02-22:
https://github.com/zephyrproject-rtos/zephyr/pull/13659

ManojSubbarao, 2019-02-28:
Closed via #13659

closed 2019-02-28 by ManojSubbarao