zephyrproject-rtos/zephyr #8098: tests/drivers/rtc/rtc_basic_api fails on Quark SE / x86

inakypg, 2018-06-01:
Toolchain: ZEPHYR_GCC_VARIANT=zephyr

Commit 43553da9b2bb9e537185695899fb5184c3b17ebe

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.0-rc2 (delayed boot 1000ms) *****
Running test suite rtc_basic_test
===================================================================
starting test - test_rtc_alarm
RTC alarm doesn't work well

    Assertion failed at /home/jenkins/wortc_alarm_callback: Invoked
RTC counter: 32814
Catch pending RTC interrupt
rkspace/zephyr-master-tcf-v0.11-branch/LABEL/verify/SHARD/1-3/ZEPHYR_GCC_VARIANT/zephyr/zephyr.git/tests/drivers/rtc/rtc_basic_api/src/test_rtc_alarm.c:115: test_rtc_alarm: test_alarm() == TC_PASS is false

FAIL - test_rtc_alarm
===================================================================
starting test - test_rtc_calendar
val_1: 33424, val_2: 66051, delta: 0:32627
PASS - test_rtc_calendar
===================================================================
===================================================================
RunID: ci-180531-1931-1708:ha9h
PROJECT EXECUTION FAILED
```

tbursztyka, 2018-06-01:
Could RTC_PRESCALER be wrong?

I thought it could be due to my dts changes but the interrupt and the relevant callback are raised/called but it's way too late. I tried with raising RTC_PRESCALER, it worked. I wonder however why the default value does not work anymore. I'll try to run a git bisect if I find a not too old commit where it used to work.

tbursztyka, 2018-06-04:
git bisect points me to: 27bdb83308301a1dbe3db9e1fbaf6560facda654