zephyrproject-rtos/zephyr #8104: open-amp: enable C library cause open-amp\open-amp\lib\common\sh_mem.c compile error

Jeepgoing, 2018-06-01:
i am designing application program. when i enabled C Library, open-amp library compile error, as below:

`/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:204:18: error: expected specifier-qualifier-list before ‘int_least8_t’
 typedef _Atomic(int_least8_t)  atomic_int_least8_t;
                  ^~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:205:18: error: expected specifier-qualifier-list before ‘uint_least8_t’
 typedef _Atomic(uint_least8_t)  atomic_uint_least8_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:206:18: error: expected specifier-qualifier-list before ‘int_least16_t’
 typedef _Atomic(int_least16_t)  atomic_int_least16_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:207:18: error: expected specifier-qualifier-list before ‘uint_least16_t’
 typedef _Atomic(uint_least16_t)  atomic_uint_least16_t;
                  ^~~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:208:18: error: expected specifier-qualifier-list before ‘int_least32_t’
 typedef _Atomic(int_least32_t)  atomic_int_least32_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:209:18: error: expected specifier-qualifier-list before ‘uint_least32_t’
 typedef _Atomic(uint_least32_t)  atomic_uint_least32_t;
                  ^~~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:210:18: error: expected specifier-qualifier-list before ‘int_least64_t’
 typedef _Atomic(int_least64_t)  atomic_int_least64_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:211:18: error: expected specifier-qualifier-list before ‘uint_least64_t’
 typedef _Atomic(uint_least64_t)  atomic_uint_least64_t;
                  ^~~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:212:18: error: expected specifier-qualifier-list before ‘int_fast8_t’
 typedef _Atomic(int_fast8_t)  atomic_int_fast8_t;
                  ^~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:213:18: error: expected specifier-qualifier-list before ‘uint_fast8_t’
 typedef _Atomic(uint_fast8_t)  atomic_uint_fast8_t;
                  ^~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:214:18: error: expected specifier-qualifier-list before ‘int_fast16_t’
 typedef _Atomic(int_fast16_t)  atomic_int_fast16_t;
                  ^~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:215:18: error: expected specifier-qualifier-list before ‘uint_fast16_t’
 typedef _Atomic(uint_fast16_t)  atomic_uint_fast16_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:216:18: error: expected specifier-qualifier-list before ‘int_fast32_t’
 typedef _Atomic(int_fast32_t)  atomic_int_fast32_t;
                  ^~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:217:18: error: expected specifier-qualifier-list before ‘uint_fast32_t’
 typedef _Atomic(uint_fast32_t)  atomic_uint_fast32_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:218:18: error: expected specifier-qualifier-list before ‘int_fast64_t’
 typedef _Atomic(int_fast64_t)  atomic_int_fast64_t;
                  ^~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:219:18: error: expected specifier-qualifier-list before ‘uint_fast64_t’`

nashif, 2018-07-20:
When C library are you using? Which toolchain? 
Looks like it works fine with newlib 2.5, not versions before that.

Jeepgoing, 2018-07-22:
this issue only happen when i enabled openamp function. if i disabled openamp function, c lib works good.@nashif

galak, 2018-12-06:
I think this might get fixed when we update OpenAMP to the new version.   See PR #11355

Jeepgoing, 2018-12-07:
Hi, @galak, I have tested your branch (galak:openamp-v2018.10),  and the question still exist, i have add <stdint.h> to <stdatomic.h>, but other errors occur, our product need <math.h>, so we must enable newlib support, hope your relay.

galak, 2019-02-01:
Which c-library, how can I reproduce this?

Jeepgoing, 2019-02-03:
@galak  newlib through menuconfig

Jeepgoing, 2019-02-03:
@galak 
example：openamp  lpc54114 board, enable libc through menuconfig.

galak, 2019-02-03:
@Jeepgoing what errors are you seeing when you do this with current tree?

Jeepgoing, 2019-02-04:
@galak  there are something undefined errors from  SDK

galak, 2019-02-05:
> @galak there are something undefined errors from SDK

can you post the errors you see currently, want to make sure i'm seeing the same thing.

Jeepgoing, 2019-02-11:
@galak  Hi, thank you for your relay, do you have replicated this issue?

galak, 2019-02-19:
@Jeepgoing if you can add a log of the failure you are seeing to the issue that would be great.

Jeepgoing, 2019-02-20:
> @Jeepgoing if you can add a log of the failure you are seeing to the issue that would be great.

`/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:204:18: error: expected specifier-qualifier-list before ‘int_least8_t’
 typedef _Atomic(int_least8_t)  atomic_int_least8_t;
                  ^~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:205:18: error: expected specifier-qualifier-list before ‘uint_least8_t’
 typedef _Atomic(uint_least8_t)  atomic_uint_least8_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:206:18: error: expected specifier-qualifier-list before ‘int_least16_t’
 typedef _Atomic(int_least16_t)  atomic_int_least16_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:207:18: error: expected specifier-qualifier-list before ‘uint_least16_t’
 typedef _Atomic(uint_least16_t)  atomic_uint_least16_t;
                  ^~~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:208:18: error: expected specifier-qualifier-list before ‘int_least32_t’
 typedef _Atomic(int_least32_t)  atomic_int_least32_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:209:18: error: expected specifier-qualifier-list before ‘uint_least32_t’
 typedef _Atomic(uint_least32_t)  atomic_uint_least32_t;
                  ^~~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:210:18: error: expected specifier-qualifier-list before ‘int_least64_t’
 typedef _Atomic(int_least64_t)  atomic_int_least64_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:211:18: error: expected specifier-qualifier-list before ‘uint_least64_t’
 typedef _Atomic(uint_least64_t)  atomic_uint_least64_t;
                  ^~~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:212:18: error: expected specifier-qualifier-list before ‘int_fast8_t’
 typedef _Atomic(int_fast8_t)  atomic_int_fast8_t;
                  ^~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:213:18: error: expected specifier-qualifier-list before ‘uint_fast8_t’
 typedef _Atomic(uint_fast8_t)  atomic_uint_fast8_t;
                  ^~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:214:18: error: expected specifier-qualifier-list before ‘int_fast16_t’
 typedef _Atomic(int_fast16_t)  atomic_int_fast16_t;
                  ^~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:215:18: error: expected specifier-qualifier-list before ‘uint_fast16_t’
 typedef _Atomic(uint_fast16_t)  atomic_uint_fast16_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:216:18: error: expected specifier-qualifier-list before ‘int_fast32_t’
 typedef _Atomic(int_fast32_t)  atomic_int_fast32_t;
                  ^~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:217:18: error: expected specifier-qualifier-list before ‘uint_fast32_t’
 typedef _Atomic(uint_fast32_t)  atomic_uint_fast32_t;
                  ^~~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:218:18: error: expected specifier-qualifier-list before ‘int_fast64_t’
 typedef _Atomic(int_fast64_t)  atomic_int_fast64_t;
                  ^~~~~~~~~~~~
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:219:18: error: expected specifier-qualifier-list before ‘uint_fast64_t’`

this is my compile error.

galak, 2019-02-20:
@Jeepgoing can you put the full build log.

Also have you tried the 0.10.0-rc3 sdk?

Jeepgoing, 2019-02-21:
> @Jeepgoing can you put the full build log.
> 
> Also have you tried the 0.10.0-rc3 sdk?

full log as below, and i have tried 0.10.0-rc3 sdk, it is the same error.

```
amine@amine-VirtualBox:~/work/zephyr/zephyrproject/zephyr/samples/subsys/ipc/openamp/lpcxpresso54114$ ninja
[0/1] Re-running CMake...
Zephyr version: 1.14.0
-- Selected BOARD lpcxpresso54114_m4
-- Loading /home/amine/work/zephyr/zephyrproject/zephyr/boards/arm/lpcxpresso54114/lpcxpresso54114_m4.dts as base
-- Overlaying /home/amine/work/zephyr/zephyrproject/zephyr/dts/common/common.dts
lpcxpresso54114_m4.dts_compiled: Warning (simple_bus_reg): /soc/gpio@0: simple-bus unit address format error, expected "4008c000"
lpcxpresso54114_m4.dts_compiled: Warning (simple_bus_reg): /soc/gpio@1: simple-bus unit address format error, expected "4008c000"
Parsing Kconfig tree in /home/amine/work/zephyr/zephyrproject/zephyr/Kconfig
Loading /home/amine/work/zephyr/zephyrproject/zephyr/samples/subsys/ipc/openamp/lpcxpresso54114/zephyr/.config as base
Configuration written to '/home/amine/work/zephyr/zephyrproject/zephyr/samples/subsys/ipc/openamp/lpcxpresso54114/zephyr/.config'
-- Cache files will be written to: /home/amine/.cache/zephyr
-- Host:    Linux/x86_64
-- Target:  Generic/arm
-- Machine: cortexm
-- C_FLAGS :  -Wall -Wextra
-- Build type:  Debug
-- Host:    Linux/x86_64
-- Target:  Generic/arm
-- Machine: cortexm
Including module(s): 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/amine/work/zephyr/zephyrproject/zephyr/samples/subsys/ipc/openamp/lpcxpresso54114
[6/118] Building C object CMakeFiles/app.dir/src/main.c.obj
In file included from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/include/openamp/virtio.h:11:0,
                 from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/include/openamp/rpmsg_virtio.h:18,
                 from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/include/openamp/open_amp.h:12,
                 from ../src/main.c:18:
zephyr/ext/hal/libmetal/libmetal/lib/include/metal/spinlock.h: In function ‘metal_spinlock_acquire’:
zephyr/ext/hal/libmetal/libmetal/lib/include/metal/spinlock.h:47:34: warning: passing argument 1 of ‘atomic_flag_test_and_set’ from incompatible pointer type [-Wincompatible-pointer-types]
  while (atomic_flag_test_and_set(&slock->v)) {
                                  ^
In file included from zephyr/ext/hal/libmetal/libmetal/lib/include/metal/atomic.h:19:0,
                 from zephyr/ext/hal/libmetal/libmetal/lib/include/metal/system/zephyr/mutex.h:19,
                 from zephyr/ext/hal/libmetal/libmetal/lib/include/metal/mutex.h:22,
                 from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/include/openamp/rpmsg.h:16,
                 from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/include/openamp/open_amp.h:11,
                 from ../src/main.c:18:
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:398:1: note: expected ‘volatile atomic_flag * {aka volatile struct <anonymous> *}’ but argument is of type ‘atomic_int * {aka struct <anonymous> *}’
 atomic_flag_test_and_set(volatile atomic_flag *__object)
 ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/include/openamp/virtio.h:11:0,
                 from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/include/openamp/rpmsg_virtio.h:18,
                 from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/include/openamp/open_amp.h:12,
                 from ../src/main.c:18:
zephyr/ext/hal/libmetal/libmetal/lib/include/metal/spinlock.h: In function ‘metal_spinlock_release’:
zephyr/ext/hal/libmetal/libmetal/lib/include/metal/spinlock.h:59:20: warning: passing argument 1 of ‘atomic_flag_clear’ from incompatible pointer type [-Wincompatible-pointer-types]
  atomic_flag_clear(&slock->v);
                    ^
In file included from zephyr/ext/hal/libmetal/libmetal/lib/include/metal/atomic.h:19:0,
                 from zephyr/ext/hal/libmetal/libmetal/lib/include/metal/system/zephyr/mutex.h:19,
                 from zephyr/ext/hal/libmetal/libmetal/lib/include/metal/mutex.h:22,
                 from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/include/openamp/rpmsg.h:16,
                 from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/include/openamp/open_amp.h:11,
                 from ../src/main.c:18:
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:406:1: note: expected ‘volatile atomic_flag * {aka volatile struct <anonymous> *}’ but argument is of type ‘atomic_int * {aka struct <anonymous> *}’
 atomic_flag_clear(volatile atomic_flag *__object)
 ^~~~~~~~~~~~~~~~~
../src/main.c: At top level:
../src/main.c:95:18: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .get_features = virtio_get_features,
                  ^~~~~~~~~~~~~~~~~~~
[... 188 lines cut ...]
                                  ^
In file included from zephyr/ext/hal/libmetal/libmetal/lib/include/metal/atomic.h:19:0,
                 from zephyr/ext/hal/libmetal/libmetal/lib/include/metal/io.h:21,
[... cut ...]
                 from /home/amine/work/zephyr/zephyrproject/zephyr/ext/lib/ipc/open-amp/open-amp/lib/rpmsg/rpmsg_virtio.c:14:
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdatomic.h:406:1: note: expected ‘volatile atomic_flag * {aka volatile struct <anonymous> *}’ but argument is of type ‘atomic_int * {aka struct <anonymous> *}’
 atomic_flag_clear(volatile atomic_flag *__object)
 ^~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
amine@amine-VirtualBox:~/work/zephyr/zephyrproject/zephyr/samples/subsys/ipc/openamp/lpcxpresso54114$
```

wjliang, 2019-03-01:
Please see libmetal patches: https://github.com/OpenAMP/libmetal/pull/85

galak, 2019-03-08:
@Jeepgoing should hopefully now be fixed with PR #14203.

Jeepgoing, 2019-03-13:
@galak ， ok, it's work, thanks.

closed 2019-03-13 by Jeepgoing