zephyrproject-rtos/zephyr #8111: kconfiglib   warning

hongshui3000, 2018-06-01:
Development Environment： windows 7

run：
cmake -GNinja -DBOARD=nrf52_pca10040 ..

Find a lot of warning information：
warning: style: quotes recommended around default value for string symbol SOC_SERIES 

kconfiglib.py code：
                    sym.kconfig._warn("style: quotes recommended around "
                                      "default value for string symbol "
                                      + _name_and_loc(sym))

ulfalizer, 2018-06-01:
@hongshui3000 
Are you running the latest version of master? Do you have any custom board definitions?

One of the `SOC_SERIES` definition locations that you see listed in the warning has a `default foo`, when it should have a default `default "foo"`. You can check out what the `default`s are supposed to look like in the [symbol reference](http://docs.zephyrproject.org/reference/kconfig/CONFIG_SOC_SERIES.html#cmdoption-arg-config-soc-series).

The warning unfortunately gets a bit spammy for symbols defined in multiple locations (like `SOC_SERIES`, which has a ton of definitions). It's a bit tricky to work around, unfortunately (you might not know the type of the symbol until you've seen all the definitions, and all of the definitions get merged into a single symbol).

The way you should read the warning is that *some* (probably just one) of those definitions has an unquoted string default (which ends up as part of the final symbol). I agree that it isn't optimal.

Later on these warnings should be caught earlier at least.

hongshui3000, 2018-06-04:
@ulfalizer The problem is solved. Thank you for your advice.

pfalcon, 2018-06-04:
@hongshui3000 : Closing then. Feel free to reopen with additional information if something goes wrong.

closed 2018-06-04 by pfalcon