zephyrproject-rtos/zephyr #8116: tests/kernel/profiling/profiling_api fails to complete on minnowboard

inakypg, 2018-06-01:
Toolchain: ZEPHYR_GCC_VARIANT=zephyr

Commit 9a7538b95d9750ecd7cc714092588f988b384847

After 60 seconds, it never completes or resets, printing the below console output. The same testcase on qemu takes less than one second.

```

>>>>SecStartup
>>>>MemoryInit Done
>>>>BdsEntry

>>>>Start boot option, Press <F2> or <DEL> to enter setup page(5 Sec).....
UEFI Interactive Shell v2.2
EDK II
UEFI v2.60 (EDK II, 0x00010000)
Mapping table
      FS0: Alias(s):HD0b0b:;BLK1:
          PciRoot(0x0)/Pci(0x14,0x0)/USB(0x1,0x0)/HD(1,GPT,B5F2229C-01EA-49AA-8ECF-9E2D4CF2C4FF,0x800,0x2CCCB)
     BLK0: Alias(s):
          PciRoot(0x0)/Pci(0x14,0x0)/USB(0x1,0x0)
Press ESC in 5 seconds to skip startup.nsh or any other key to continue.Press ESC in 4 seconds to skip startup.nsh or any other key to continue.Press ESC in 3 seconds to skip startup.nsh or any other key to continue.Press ESC in 2 seconds to skip startup.nsh or any other key to continue.Press ESC in 1 seconds to skip startup.nsh or any other key to continue.
Shell> fs0:\EFI\BOOT\bootx64
TCF Booting kernel-0f9a19103a.elf from root device hd0,gpt1

WARNING: no console will be available to OS

error: no suitable video mode found.

***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.0-rc2 (delayed boot 1000ms) *****
Running test suite profiling_api
===================================================================
starting test - test_call_stacks_analyze_main
from main thread:
Test (real size 1024):	unused 772	usage 252 / 1024 (24 %)
Test (real size 1024):	unused 836	usage 188 / 1024 (18 %)
Test (real size 512):	unused 488	usage 24 / 512 (4 %)
Test (real size 512):	unused 176	usage 336 / 512 (65 %)
PASS - test_call_stacks_analyze_main
===================================================================
starting test - test_call_stacks_analyze_idle
from idle thread:
PASS - test_call_stacks_analyze_idle
===================================================================
starting test - test_call_stacks_analyze_workq
from workq:
Test (real size 1024):	unused 768	usage 256 / 1024 (25 %)
Test (real size 1024):	unused 632	usage 392 / 1024 (38 %)
Test (real size 512):	unused 320	usage 192 / 512 (37 %)
Test (real size 512):	unused 176	usage 336 / 512 (65 %)
Test (real size 1024):	unused 768	usage 256 / 1024 (25 %)
Test (real size 1024):	unused 632	usage 392 / 1024 (38 %)
Test (real size 512):	unused 320	usage 192 / 512 (37 %)
Test (real size 512):	unused 176	usage 336 / 512 (65 %)
PASS - test_call_stacks_analyze_workq
===================================================================
starting test - test_k_thread_foreach
```

SavinayDharmappa, 2018-06-08:
CC @ramakrishnapallala 
After debugging the issue we found that the timeouts are not happening as expected on minnowboard.

In this test, test_k_thread_foreach sleeps for 1 msec and should resume after 1msec timeout happens and continue the test but the system is stuck in ilde loop.

ManojSubbarao, 2018-06-11:
@SavinayDharmappa  what is the status?

SavinayDharmappa, 2018-06-21:
The timer value configured in _timer_idle_enter was not taking into account of current value of counter because of which any time-out below 10 ticks were not  happening. But when next timeout expiry is less the CONFIG_TICKLESS_IDLE_THRESH, there won't any configuration of timer value. In this case still system is not able to wakeup from cpu idle. I am still debugging the issue.

SavinayDharmappa, 2018-10-24:
can u test it again. Issue may not exsist as reported...as test case related to it is being moved

nashif, 2019-01-30:
not relevant or tested on this platform anymore.

closed 2019-01-30 by nashif