zephyrproject-rtos/zephyr #8117: tests/kernel/errno crashes on minnowboard

inakypg, 2018-06-01:
Toolchain: ZEPHYR_GCC_VARIANT=zephyr

Commit f90416c680d19170baf4eb3a3d716f37bc5b124b

```
>>>>SecStartup
>>>>MemoryInit Done
>>>>BdsEntry

>>>>Start boot option, Press <F2> or <DEL> to enter setup page(5 Sec).....
UEFI Interactive Shell v2.2
EDK II
UEFI v2.60 (EDK II, 0x00010000)
Mapping table
      FS0: Alias(s):HD0b0b:;BLK1:
          PciRoot(0x0)/Pci(0x14,0x0)/USB(0x1,0x0)/HD(1,GPT,19BF1E70-3CB6-4304-9893-EEF1E04456AC,0x800,0x2CCCB)
     BLK0: Alias(s):
          PciRoot(0x0)/Pci(0x14,0x0)/USB(0x1,0x0)
Press ESC in 5 seconds to skip startup.nsh or any other key to continue.Press ESC in 4 seconds to skip startup.nsh or any other key to continue.Press ESC in 3 seconds to skip startup.nsh or any other key to continue.Press ESC in 2 seconds to skip startup.nsh or any other key to continue.Press ESC in 1 seconds to skip startup.nsh or any other key to continue.
Shell> fs0:\EFI\BOOT\bootx64
TCF Booting kernel-6f7dcbc59b.elf from root device hd0,gpt1

WARNING: no console will be available to OS

error: no suitable video mode found.

***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.0-rc2 (delayed boot 1000ms) *****
***** CPU Page Fault (error code 0x00000000)
Supervisor thread read address 0x00000000
PDE: 0x027 Present, Writable, User, Execute Enabled
PTE: 0x00 Non-present, Read-only, Supervisor, Execute Enabled
***** Stack Check Fail! *****
Current thread ID = 0x0010b080
eax: 0x0011f240, ebx: 0x0011b0a8, ecx: 0x00101f18, edx: 0x00000043
esi: 0x001077f1, edi: 0x00000000, ebp: 0x0011b000, esp: 0x0011b000
eflags: 0x00210216 cs: 0x0008
eip: 0x00101559
Fatal fault in essential thread! Spinning...
```

SavinayDharmappa, 2018-06-06:
I am not reproduce on my board locally, below is the log,

***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.12.0-rc2 (delayed boot 1000ms) *****
Running test suite context_errno
===================================================================
starting test - test_thread_context
PASS - test_thread_context
===================================================================
===================================================================
PROJECT EXECUTION SUCCESSFUL

inakypg, 2018-06-06:
Confirmed it stopped failing in our environment around 6/5 8am PDT time, last failure was on commit 

Commit of last failing build c5cb8e943c40e997df3794ee2911868ee37f231a
Commit in first passing build 59bf65f481205c0171a0665845f7eb520fa65a56

MaureenHelm, 2018-06-06:
@inakypg I don't see how the passing commit would have fixed this test. Are you confident that this issue is resolved?

inakypg, 2018-06-06:
Sorry @MaureenHelm, meant that it was fixed between last failing and first failing. We'd have to do a full bisect to identify who fixed it.

Our CI doesn't run for every commit, but for whichever commit is available when it is ready to start again.

nashif, 2018-06-08:
nothing between those 2 commits changes anything related, all changes are in unrelated tests/samples. I am closing this now, if it happens again, please reopen.

closed 2018-06-08 by nashif

inakypg, 2018-06-08:
I keep seeing it, failing consistently--at this point we could argue it can be a hardware problem

```
E#1      @srrsotc05/minnowboard-34  console output: ***** delaying boot 1000ms (per build configuration) *****
E#1      @srrsotc05/minnowboard-34  console output: ***** Booting Zephyr OS 1.12.0-rc2 (delayed boot 1000ms) *****
E#1      @srrsotc05/minnowboard-34  console output: ***** CPU Page Fault (error code 0x00000000)
E#1      @srrsotc05/minnowboard-34  console output: Supervisor thread read address 0x00000000
E#1      @srrsotc05/minnowboard-34  console output: PDE: 0x027 Present, Writable, User, Execute Enabled
E#1      @srrsotc05/minnowboard-34  console output: PTE: 0x00 Non-present, Read-only, Supervisor, Execute Enabled
E#1      @srrsotc05/minnowboard-34  console output: ***** Stack Check Fail! *****
E#1      @srrsotc05/minnowboard-34  console output: Current thread ID = 0x0010b080
E#1      @srrsotc05/minnowboard-34  console output: eax: 0x0011f240, ebx: 0x0011b0a8, ecx: 0x00101f18, edx: 0x00000043
E#1      @srrsotc05/minnowboard-34  console output: esi: 0x00107811, edi: 0x00000000, ebp: 0x0011b000, esp: 0x0011b000
E#1      @srrsotc05/minnowboard-34  console output: eflags: 0x00210216 cs: 0x0008
E#1      @srrsotc05/minnowboard-34  console output: eip: 0x00101559
E#1      @srrsotc05/minnowboard-34  console output: Fatal fault in essential thread! Spinning...
```

@SavinayDharmappa, can you try in this very specific board?