zephyrproject-rtos/zephyr #8125: About BMI160 reading issue.

Harishsam07, 2018-06-02:
Dear all
we are trying to interface BMI160 sensor with ARM controller through I2C protocol but we are not getting proper reading from sensor we need basic initialize register setting of BMI160.
standard reading format should be value  0.0 m/s^2  0.0 m/s^2  9.8 m/s^2 
but we getting 
-1.8183590m/s^2  Ay=-6.0849610m/s^2  Az=-0.6469730m/s^2

punitvara, 2018-06-04:
@Harishsam07  Which ARM board ? Can you add specific details please ?

Harishsam07, 2018-06-05:
we are using ARM Controller LPC2138 and I2C protocol.  now we facing issue accelerometer reading are not getting refresh.

Harishsam07, 2018-06-08:
Currently we have set following initialization parameter but accelerometer reading are not getting refresh is constant while changing the position of BMI160. 

        i2c_write_data(0x7E,0xB6);                       //Soft reset to clean register
	delay(100);	
	i2c_write_data(0x7E,0xB0);                       //clear all
	delay(100);	
	
	i2c_write_data(0x46,0x00);                       //FIFO Disable
	delay(10);
	i2c_write_data(0x47,0x00);                       //FIFO Disable
	delay(10);

	i2c_write_data(0x70,0x06);                       //Enable Only I2C Disable SPI,Disable Watchdog timer.
	delay(10);
				
	i2c_write_data(0x77,0x00);                        //Offset disable
	delay(10);

	i2c_write_data(0x43,0x03);	                //GYRO_RAGE -> +- 250°/s : 7.6 m°/LSB
	delay(10);
	i2c_write_data(0x41,0x0C);	                //ACC_RANGE => +- 16g
	delay(10);
	i2c_write_data(0x40,0x2C);	                // ACC_CONF data rate : 1600Hz
	delay(10);
	i2c_write_data(0x42,0x2C);	                // GYRO_CONF data rate : 1600Hz
	delay(10);
	
	i2c_write_data(0x7E,0x11);	//Power up Accelerometer
	delay(10);
	i2c_write_data(0x7E,0x15);  //Power up Gyroscope
	delay(10);

Harishsam07, 2018-08-30:
I have solved my isuue successfuly..

closed 2018-08-30 by Harishsam07

marcoantonioara, 2019-07-02:
BMI160 has driver to SPI in Zephyr. I have this sensor connected in a board using the i2c. 
Please share your code with us