zephyrproject-rtos/zephyr #8145: samples/subsys/usb/dfu: Build failure, Reference to non-existent node or label "slot0_partition"

spoorthik, 2018-06-04:
Test case: USB DFU sample at samples/subsys/usb/dfu

Issue: Fails to build, error states "Reference to non-existent node or label "slot0_partition""

Execution log:
```
-- Found PythonInterp: /bin/python3 (found suitable version "3.5.4", minimum required is "3.4")
-- Selected BOARD quark_se_c1000_devboard
Zephyr version: 1.12.0
Parsing Kconfig tree in /home/kspoorth/work/latest_zephyr/Kconfig
Using /home/kspoorth/work/latest_zephyr/boards/x86/quark_se_c1000_devboard/quark_se_c1000_devboard_defconfig as base
Merging /home/kspoorth/work/latest_zephyr/samples/subsys/usb/dfu/prj.conf
-- Generating zephyr/include/generated/generated_dts_board.h
ERROR (path_references): Reference to non-existent node or label "slot0_partition"

ERROR: Input tree has errors, aborting (use -f to force output)
CMake Error at /home/kspoorth/work/latest_zephyr/cmake/dts.cmake:83 (message):
  command failed with return code: 2
Call Stack (most recent call first):
  /home/kspoorth/work/latest_zephyr/cmake/app/boilerplate.cmake:268 (include)
  CMakeLists.txt:1 (include)

-- Configuring incomplete, errors occurred!
```

Steps to reproduce:
cd zephyr.git/samples/subsys/usb/dfu
mkdir build && cd build
cmake -DBOARD=quark_se_c1000_devboard .. 

Tried the same steps for qemu_x86 and found same error.

Latest commit in which failure observed: 7a18b083ebe4712fbb258d09c0f53a71b7f46678

jfischer-phytec-iot, 2018-06-04:
quark_se_c1000_devboard.dts has no partition layout for a flash device.

nashif, 2018-06-04:
this sample is whitelisted for nrf52840_pca10056 only, so it is not expected to work on anything else unless we have the partitions layout in place.

closed 2018-06-04 by nashif