zephyrproject-rtos/zephyr #8150: Doc: Update Zephyr security overview

erwango, 2018-06-04:
http://docs.zephyrproject.org/security/security-overview.html would deserve an update.

"Current Security definition" is out-of-date: 
- `Memory protection and thread separation was added in version 1.10.0 for X86. ARM and ARC support is targeted for version 1.11.0.`
- `Memory protection and task separation techniques are in scope starting with version 1.10.0.`
- and so on...

nashif, 2018-06-04:
@agross-linaro I remember we talked about it a while back, can you please update?

lpereira, 2018-06-04:
FWIW, I've been adding things to a wiki page describing [security best practices](https://github.com/zephyrproject-rtos/zephyr/wiki/BestPracticesSecurity).  Eventually this kind of stuff should go into the documentation.