zephyrproject-rtos/zephyr #8160: BUILD_ASSERT doesn't work outside gcc

andyross, 2018-06-04:
As discovered in #8093, our generic implementation of BUILD_ASSERT produces preprocessor errors always, not just when the assertion case is failing.

We don't actually use this on any platforms that don't already support C11's _Static_assert(), so this is fairly low priority.  But someone should look at it at some point.

andyross, 2019-03-05:
Closing this one.  We have toolchain support for this already in non-gcc toolchains, and no gcc ones known are going to hit the "no _Static_assert() available" case anyway.

closed 2019-03-05 by andyross

rgundi, 2019-03-05:
@andyross XCC currently doesn't have _Static_assert() defined, but the generic BUILD_ASSERT defined in include/toolchain/common.h works just fine. I see that the error like below is generated only when the assertion case is failing.
warning: division by zero
error: enumerator value for ‘__build_assert15’ is not an integer constant