zephyrproject-rtos/zephyr #8168: Need for buffered printk() output

pfalcon, 2018-06-04:
There were multiple experiences and reports from different people that enabling debug logging affects the behavior of non-trivial (e.g. time-sensitive code). That's because debug logging happens via printk(), which (by default) uses polled UART output. For typical 115200 UART speed, that means 11.5KB/s throughput, and then even for 10 char output, 1ms is spent in polling, and most of logging messages are times longer. Overall, such logging affects real-time code noticeably, usually masking any concurrency and subtle-timing issues.

So, we'd need to implement some kind of buffered output to alleviate this issue. There're 2 approaches to this:

1. Implement interrupt-driven UART for printk, so logging output hopefully takes negligible (for many uses) amount of CPU time.
2. Implement an in-memory buffer for output, with some adhoc support for flushing to the underlying output device (usually still UART), e.g. on buffer becoming full or using an API call a user will need to sprinkle in the code at appropriate places to get the output.

Both approaches has benefits and drawbacks:

1. Requires interrupt-driven UART, which is not supported on all platforms. One big benefit is that the output is still "realtime" (immediately visible to the user). Of course, there's a concern that on buffer overflow it'll stop being so, but that can be handled by configuring buffer size (to an arbitrary value, include tens and hundreds of KBs). This method still has non-zero overhead, so might affect tight realtime code.

2. The output is not visible to the user by default, so it's "collect and stop the world" approach. As such, it'll be *much* harder to use. At the same time, it's more compatible and allows to achieve absolutely minimal overhead during the "collection" phase.

pfalcon, 2018-06-04:
@nashif, @mike-scott, @jukkar, @tbursztyka, @rlubos: Comments are welcome. The main usecase for me is debugging IP stack, where each of us saw enabling logging skew the outcome of the debugging session, up to complete masking of the issues seen otherwise.

mike-scott, 2018-06-04:
I hit this issue *all the time*.  I was dealing with this while looking at the scheduler.

Certainly a useful issue to track / work on.

It can be quite complex:
- panic / aborts need to flush the buffer
- buffer overflows in critical portions of the code(?)
- multi-threaded behavior makes the buffered output look odd
- etc.

nashif, 2018-06-04:
isn't this the same as https://github.com/zephyrproject-rtos/zephyr/issues/6186 and the logging system/printk overhaul?

pfalcon, 2018-06-05:
> isn't this the same as #6186 and the logging system/printk overhaul?

Well, this issue is definitely not about "overhaul" or (as #8023 linked from #6186) "major redesign". Those tickets definitely affect this issue, though looking at #8023, I'm not sure if it's ready for prime time.

So, it's mostly to collect opinions from folks to @mike-scott and @jukkar who suffer from this issue daily, to see if they think that even partial/interim solution would be helpful (or maybe they have some hack already ;-) ).

jukkar, 2018-06-05:
I am personally waiting #6186 to finish, then we can start to look if there is anything else to do to make debug prints usable.

nashif, 2018-06-05:
> I'm not sure if it's ready for prime time.

not sure what that means. This is planned for 1.13 and we have this already submitted and being reviewed before even 1.13 has started. Sure there is still lot to do, but the skeleton and framework for adding more backends will be merged soonish.

So, I am not sure what you are suggesting here, do you have another solution to all of this that does not conflict with the efforts being done in #6186?

pfalcon, 2018-06-05:
@nashif 

> not sure what that means. This is planned for 1.13 and we have this already submitted and being reviewed before even 1.13 has started.

I see that #8023 is the only proposed solution for #6186. And even from cursory look I gave to it, I found a bunch of smallish issues which need to be fixed before it can be merged. And that's without talking about higher-level design and architecture questions. (Which I'm sure were discussed already, and concerns with the design were, well, rejected. But that doesn't mean they aren't valid).

> So, I am not sure what you are suggesting here, do you have another solution to all of this that does not conflict with the efforts being done in #6186?

Well, this is definitely an RFC. It was a long-time concern that logging affects network debugging, and somehow just now it should be clear that we can't have good progress in debugging non-deterministic issues without improved debug output support. So, this issue is to ask what other people think or maybe even do about it.

"Sit and wait for #6186" is already a good response. I have to admit that I'm not much thrilled by it, because it's either typical "sit and wait" approach (which can go for years), or "go into levels of recursion - when you want to debug X, you need to debug Y, Z, ... etc. first". So, I'm keen to explore possibilities while #6186 is in works (that also has added benefit of better understanding the architecture we have - we have layers, what each of them can do?)

nashif, 2018-06-05:
`Need for buffered printk() output` is exactly what we are trying to do in #8023 and for the reasons you mention above and it is WIP with a proposed design and a proposed implementation. If you want to explore other ways, that is up to you, however this will conflict with work in progress already, IMO this will be a hard-conflict, because we are changing how printk works completely. So my suggestion is to join the effort instead of starting a new one in parallel and provide your input and hopefully code as well to make this go faster.

pfalcon, 2018-06-05:
> because we are changing how printk works completely

Yes, you do, and I'm not sure in which way (missed RFC on the mailing list apparently) and why (though all questions like that nowadays can be answered: because of kernel/userspace separation). That's the feedback I can provide, and as that's not much, I'm indeed keen to explore ways to add buffering to printk *as it is now*. That can be a feedback too - for example, if something could be done easy enough now, and won't be so later, when you change it completely. (Otherwise, I already feel sorry for opening this issue ;-), and indeed, I started to submit "alternative issues" to my fork at https://github.com/pfalcon/zephyr/issues)

nashif, 2018-06-05:
#6186 has been open for almost 4 months now, it is marked as a `feature`, has been on the roadmap since it was opened, has been discussed on the TSC in multiple occasions and also was mentioned on the mailing list multiple times, i.e. the intention was there. This is open issue is fine and I am not saying you should not have opened it, you can't know everything. But now that you are in the know, it would make sense to join the discussion and not duplicate.

This btw has nothing to do with userspace. It is for the same reasons you mention above. Networking is not the only subsystem that suffers under that and TBH, logging and usage of printk in Zephyr is below sub-optimal, so it is more than just being able to debug 1 single bug.

pfalcon, 2018-06-05:
So, with the above in mind, I'd like to survey ways to add buffering to printk (and only printk, skipping any "logging" talks) now.

1. First of all, there's already a kind of buffering in printk() - to avoid userspace/kernel switching overhead, done by @andrewboie in 756f907274. With some effort, it could be hacked into no.2 solution ("collect and stop-the-world").
2. Using subsys/console/. console_putchar could be passed to `__printk_hook_install`, giving solution no.1 (interrupt-driven output), voila. (Well, there're complications of different prototypes between console_putchar and `__printk_hook_install` arg, and handling buffer overflow). We discussed hooking printk into console subsys with @andrewboie some time ago, and more recently https://github.com/zephyrproject-rtos/zephyr/pull/6815. (I was skeptical about such hooking, yeah. I may imagine p.1 solution above was made because printk/console-subsys discussion didn't progress at that time.)

Oh, and yeah, while we agreed not to talk about logging, but besides printk hook, even with the current logging subsys, we have logging hook enabled by CONFIG_SYS_LOG_EXT_HOOK. Then we have https://github.com/zephyrproject-rtos/zephyr/tree/master/tests/subsys/logging/logger-hook which does implement circular buffer for logging, except as a test. That "can" could be refactored as a core component, and we'd have solution no.2.

pfalcon, 2018-06-05:
> But now that you are in the know, it would make sense to join the discussion 

I believe I submitted more comments for #8023 than anyone before so far, so I definitely joined the discussion ;-).

> and not duplicate.

I don't see that as "duplication", but rather as "exploring missed opportunities". Yes, it looks like it's being done a bit too late, but as they say, better late than never.

pfalcon, 2018-10-09:
With new logger having got in, and so much effort went into (well, still going, logger/shell migration is probably going to be major theme of 1.14, with little of anything else done), let's hope that it indeed fixes this issue too, so let me close it.

closed 2018-10-09 by pfalcon