zephyrproject-rtos/zephyr #8172: Networking tests failing with an assertion on frdm_k64f

MaureenHelm, 2018-06-05:
The following networking tests are failing with an assertion on frdm_k64f:

tests/net/vlan/net.vlan
```
***** Booting Zephyr OS v1.12.0-rc2-75-g7a18b08 *****                           
Running test suite net_vlan_test
===================================================================
starting test - test_vlan_setup

    Assertion failed at /home/maureen/zephyr/tests/net/vlan/src/main.c:337: test
Invalid numer of VLANs 8 vs 4

FAIL - test_vlan_setup
```

tests/net/checksum_offload/net.offload
```
***** Booting Zephyr OS v1.12.0-rc2-75-g7a18b08 *****                           
Running test suite net_chksum_offload_test
===================================================================
starting test - eth_setup

    Assertion failed at /home/maureen/zephyr/tests/net/checksum_offload/src/main
Invalid number of interfaces (3 vs 2)

FAIL - eth_setup
```

A full regression with device testing was run on frdm_k64f on commit 7a18b083ebe4712fbb258d09c0f53a71b7f46678:
```
$ sanitycheck -p frdm_k64f --device-testing --device-serial /dev/ttyACM0 
Cleaning output directory /home/maureen/zephyr/sanity-out
Building testcase defconfigs...
256 tests selected, 55151 tests discarded due to filters
total complete:   27/ 256  10%  failed:    0

frdm_k64f                 samples/mpu/mpu_stack_guard_test/test              FAILED: timeout
	see: sanity-out/frdm_k64f/samples/mpu/mpu_stack_guard_test/test/handler.log

total complete:   46/ 256  17%  failed:    1

frdm_k64f                 samples/mpu/mpu_stack_guard_test/test_stack_guard  FAILED: timeout
	see: sanity-out/frdm_k64f/samples/mpu/mpu_stack_guard_test/test_stack_guard/handler.log

total complete:  112/ 256  43%  failed:    2

frdm_k64f                 tests/net/vlan/net.vlan                            FAILED: failed
	see: sanity-out/frdm_k64f/tests/net/vlan/net.vlan/handler.log

total complete:  126/ 256  49%  failed:    3

frdm_k64f                 samples/drivers/watchdog/test                      FAILED: timeout
	see: sanity-out/frdm_k64f/samples/drivers/watchdog/test/handler.log

total complete:  144/ 256  56%  failed:    4

frdm_k64f                 tests/net/utils/net.util                           FAILED: timeout
	see: sanity-out/frdm_k64f/tests/net/utils/net.util/handler.log

total complete:  158/ 256  61%  failed:    5

frdm_k64f                 tests/net/icmpv6/net.icmpv6                        FAILED: timeout
	see: sanity-out/frdm_k64f/tests/net/icmpv6/net.icmpv6/handler.log

total complete:  159/ 256  62%  failed:    6

frdm_k64f                 tests/benchmarks/sys_kernel/benchmark.kernel       FAILED: failed
	see: sanity-out/frdm_k64f/tests/benchmarks/sys_kernel/benchmark.kernel/handler.log

total complete:  173/ 256  67%  failed:    7

frdm_k64f                 tests/net/checksum_offload/net.offload             FAILED: failed
	see: sanity-out/frdm_k64f/tests/net/checksum_offload/net.offload/handler.log

total complete:  196/ 256  76%  failed:    8

frdm_k64f                 tests/net/lib/http_header_fields/net.http          FAILED: timeout
	see: sanity-out/frdm_k64f/tests/net/lib/http_header_fields/net.http/handler.log

total complete:  207/ 256  80%  failed:    9

frdm_k64f                 tests/net/trickle/net.trickle                      FAILED: timeout
	see: sanity-out/frdm_k64f/tests/net/trickle/net.trickle/handler.log

total complete:  208/ 256  81%  failed:   10

frdm_k64f                 tests/kernel/fifo/fifo_api/kernel.fifo.poll        FAILED: timeout
	see: sanity-out/frdm_k64f/tests/kernel/fifo/fifo_api/kernel.fifo.poll/handler.log

total complete:  222/ 256  86%  failed:   11

frdm_k64f                 tests/net/lib/mqtt_packet/net.mqtt.packet          FAILED: timeout
	see: sanity-out/frdm_k64f/tests/net/lib/mqtt_packet/net.mqtt.packet/handler.log

total complete:  235/ 256  91%  failed:   12

frdm_k64f                 tests/drivers/adc/adc_simple/peripheral.adc        FAILED: failed
	see: sanity-out/frdm_k64f/tests/drivers/adc/adc_simple/peripheral.adc/handler.log

total complete:  256/ 256  100%  failed:   13
243 of 256 tests passed with 0 warnings in 7090 seconds
```

nashif, 2018-06-05:
If I am not mistaken, those tests are designed to run in loopback mode only and never actually worked on device with real network driver, I have seen them before but was focusing on kernel tests, we need to fix the tests to run on real hw.

jukkar, 2018-06-05:
The reason for this error is that as frdm-k64f has an ethernet port, the vlan test application gets confused by this and reports an error, as normally it is run in self contained unit.

commit 59bf65f4 (2018-06-05): tests: net: Fix tests so they can be run in real hw

Both the vlan and checksum_offload tests expect to be run in
controlled environment where the test application is providing
ethernet connectivity. If the board under test has also ethernet
support, then internal checks will fail in the test application.
Because of this, disable on board ethernet so that the tests will
pass.

Fixes #8172