zephyrproject-rtos/zephyr #8181: Samples should only be enabling their dependencies, not their dependencies dependencies

SebastianBoe, 2018-06-05:
The current Kconfig tooling is not able to resolve option dependencies.

This means that if an app uses a feature it can't simply express this with

CONFIG_FEATURE=y

in it's prj.conf, it also needs to also enable all of the feature's dependencies. e.g.

CONFIG_FEATURE=y
CONFIG_FEATURE_DEP_1=y
CONFIG_FEATURE_DEP_2=y

and the dependencies dependencies and so on.

This makes the configuration system difficult to use because the user must manually traverse the dependencies bottom-up and enable all dependencies.

It also makes it difficult to maintain samples as the samples need long lists of configuration information that is not specific to the sample, but rather the sample's dependencies.

Other DAG-based dependency systems, like package managers, and build systems, are able to transitively enable dependencies, so I don't see why Kconfig could not be adapted to support this as well.

ulfalizer, 2018-06-05:
@SebastianBoe 
This is a common request, but gets really tricky in practice.

The issue is that dependencies can be arbitrary expressions. If someone enables a `bool` symbol with unsatisfied dependencies like `depends on FOO || BAR || (BAZ && QAZ="val")` or `depends on !BAZ`, it's hard to know what they intended (can imagine that they set `BAZ=y` earlier in that second example as well).

The generality kinda restricts you to guessing in many cases. Maybe you could do it for some "obvious" cases and bail out otherwise, but I think that might make it even more confusing once you run into non-obvious cases.

SebastianBoe, 2018-06-05:
I acknowledge that this is not easy. But IMHO it is important enough that
we should attempt it despite not knowing for sure we will succeed.

"depends on FOO || BAR || (BAZ && QAZ="val")"

The FOO and BAR dep. I assume is easy here.

As for the value-based problem (QAZ="val"). I am not very familiar with the SAT problem,
but if we working with values is difficult then perhaps we could limit ourselves to a subset
 of Kconfig, E.g.:

"depends on FOO || BAR || (BAZ && QAZ_VAL)"

SebastianBoe, 2018-06-05:
> The generality kinda restricts you to guessing in many cases. Maybe you could do it for some "obvious" cases and bail out otherwise, but I think that might make it even more confusing once you run into non-obvious cases.

I'm not following you here, please do elaborate.

ulfalizer, 2018-06-05:
> The FOO and BAR dependency I assume is easy here.

The problem is that you don't know which one of them to enable. Does the user want `FOO` to be enabled to satisfy the dependency, or `BAR` (or both)? Might make a big difference in some cases.

Lots of other tricky dependencies too, like `depends on STR != "can't be this"`, and `depends on NUM > 37`.

Kconfig is definitely SAT-solver-unfriendly. Seen some projects crop up, but they tend to give up pretty quickly. :)

ulfalizer, 2018-06-05:
There isn't any fancy SAT stuff going on internally btw, just (recursive) evaluation of symbols.

The only slightly tricky part is that symbol values must be cached (and invalidated), or you go into exponential times for some stuff.

ulfalizer, 2018-06-05:
One thing I've been thinking of is generating a link to the online symbol reference in the warning. That would probably help people find it. Pretty easy to figure out the dependencies there.

SebastianBoe, 2018-06-06:
> The problem is that you don't know which one of them to enable. Does the user want FOO to be enabled to satisfy the dependency, or BAR, or BAZ && QAZ="val" (or more than one of them)? Might make a big difference in some cases.

When there are multiple solutions then I imagine the resolver should either pick one (perhaps based on a new 'suggests' property) or ask the user to choose. Both use-cases should be supported and it will be up to the user whether he wants the speed of having it resolved quickly, or the safety of seeing what decisions need to be made.

ulfalizer, 2018-06-06:
@SebastianBoe 
Suspect it might get really complex in practice, which could lead to gotchas and behavior people didn't expect.

One additional complication is that `depends on` is mostly syntactic for adding dependencies to properties. You can have separate dependencies on `prompt`s, `default`s, `select`s, etc., all on the same symbol.

What about using some heuristics to try to infer what symbol(s) the user probably wanted to enable, and print a hint along with the warning? I agree that the warning is kind of a "dead end" at the moment. Even more so for people who don't know about the symbol reference or how Kconfig works in general.

ulfalizer, 2018-06-06:
To be fair though, it's the prompt dependencies that determine whether the symbol is assignable, so should be enough to look at that (still need to take into account that a symbol can have multiple prompts if it's defined in multiple locations, but it's not horrible). Still a ton of complications elsewhere though with automatically figuring out what to enable. Think explicit might be best, but perhaps with hinting.

SebastianBoe, 2018-06-06:
> One thing I've been thinking of is having a link in the warning that goes directly to the page for the symbol in the online symbol reference. That would probably help people find it. Pretty easy to figure out the dependencies there.

> What about using some heuristics to try to infer what symbol(s) the user probably wanted to enable, and print a hint along with the warning? I agree that the warning is kind of a "dead end" at the [...]

> To be fair though, it's the prompt dependencies that determine whether the symbol is assignable, so should be enough to look at that (still need to take into account that a symbol can have multiple [...]

Improving the error feedback of the current system is welcome, but it would not resolve the two major issues that this feature request attempts to resolve, and should therefore be treated as an independent effort.

ulfalizer, 2018-06-06:
I've improved the warning now at least. Should make it much easier for new people to find the symbol reference in particular.

SebastianBoe, 2018-06-07:
That does not seem mysterious to me. Nor does it seem like a problem that would be introduced
by this effort, in fact, I suspect it will occur less frequently than now.

Assuming this Kconfig and a user that wants FOO and doesn't care about BAZ:

```
config FOO
depends on !BAZ

config BAZ
default y
```

the requirements fragment would look like this:
CONFIG_FOO=y

and the normal fragment would look like this:
CONFIG_FOO=y
CONFIG_BAZ=n

If the Kconfig changes, perhaps like this:

```
config FOO
depends on BAZ
```

then the 'requirements fragment' will still be able to produce a valid configuration but the 'normal fragment' will not, because it is assuming that FOO requires BAZ to be unset.

In this situation, the requirements fragment is much more resilient to change than the normal fragment.

In other situations, where there exists no solution that can satisfy CONFIG_FOO=y then failing is the desired behaviour.

ulfalizer, 2018-06-07:
@SebastianBoe 
I'm thinking of more advanced dependencies.

There are lots of cases where it's impossible to automatically satisfy dependencies. If you try to automatically enable something with an unsatisfied dependency like `depends on FOO != "bar"` for example, you would have to automatically come up with some value for `FOO` that's not "bar", without knowing anything about what `FOO` means.

There's tricky stuff like that all over in practice, and it'd be enough for one thing like that to get into the dependency chain for the magic to break. When it breaks, it would probably look very confusing. Instead of being helpful, I think it might make things more cryptic and fragile.

Guessing what the user wants in cases like `depends on FOO || BAR || BAZ` is really risky too I think. I think it's too magic.

For something less flexible like package management, it makes more sense.

ulfalizer, 2018-06-07:
It's a good idea to make sure that you don't have more symbols than you need too. I've seen some cases in other projects where "dummy" `menuconfig` symbols (that no code actually looked at) were used just to create menus. That makes configuration more awkward. Plain menus would be better there.

SebastianBoe, 2018-06-07:
> depends on FOO != "bar"

I'm not sure how to deal with this dependency.

I imagine one could either disallow this construct, by manually porting it to

`depends on ! FOO_VAL_BAR
`

or automatically transform it.

> There's tricky stuff like that all over in practice, and it'd be enough for one thing like that to get into the dependency chain for the magic to break. When it breaks, it would probably look very confusing. Instead of being helpful, I think it might make things more cryptic and fragile.

Noted.

> Guessing what the user wants in cases like depends on FOO || BAR || BAZ is really risky too I think. I think it's too magic.

There is no magic and there is no guessing. If the user expresses that he doesn't care about whether FOO, BAR, or BAZ is set then the user can not be surprised by any one of them being set. If the user wants to be consulted about which option should be set, then he can instruct the tool to prompt. As described earlier:

> When there are multiple solutions then I imagine the resolver should either pick one (perhaps based on a new 'suggests' property) or ask the user to choose. Both use-cases should be supported and it [...]

ulfalizer, 2018-06-07:
> I'm not sure how to deal with this dependency.

Trust me that no amount of rewriting or heuristics would allow you to automatically solve all cases in a reasonable way at least, without mind reading.

> There is no magic and there is no guessing. If the user expresses that he doesn't care about whether FOO, BAR, or BAZ is set then the user can not be surprised by any one of them being set. If the user wants to be consulted about which option should be set, then he can instruct the tool to prompt.

I would be surprised. Better to be explicit and pick one. That way you can tell what's being enabled when reading prj.conf too, which is nice.

I appreciate trying to make things easier for users, but I think this might hurt more than it helps.

SebastianBoe, 2018-06-07:
> I think the only "silver bullet" is to try to simplify Kconfig files as much as possible, have reasonable defaults, etc.

That would not resolve the two major issues that this feature request attempts to resolve, and should therefore be treated as an independent effort.

ulfalizer, 2018-06-07:
> I agree, it is better to be explicit for most users. And it is easier for a user to be explicit by having the resolver ask the user disambiguating questions than for the user to manually traverse the dependency graph.

Unless you automatically update prj.conf, it might have to ask every time `zephyr/.config` gets rebuilt, which seems a bit annoying.

It should be possible to add a hint for simple cases though, like "Did you forget to enable FOO?"

Even then, you should probably encourage people to check the docs to make sure that that's what they want.

ulfalizer, 2018-06-07:
> Correct. I imagine users want their disambiguating input saved, for instance in prj.conf.

Think it might be best to have them add it themselves, so that it stays organized.

Some hints should be doable at least. Could even include the suggested assignment(s) in them.

ulfalizer, 2018-06-07:
Should be pretty easy to point out `bool` symbols that absolutely must be enabled at least. Could split it up the dependencies into `&&` operands and check each one, printing something like "... requires enabling FOO (with CONFIG_FOO=y)" whenever it's just a symbol.

You might need to do more stuff than that in some cases, but it'd point out stuff that must be done at least.

ulfalizer, 2019-03-07:
This is ridiculously hard to fix in practice, without dramatically changing how Kconfig works. Closing for now.

SebastianBoe, 2019-03-07:
IMHO feature requests should only be closed if they are invalid. This is valid, it just requires more resources than we have at the moment.