zephyrproject-rtos/zephyr #8189: lwm2m: Quickly running out of resources when using observe

therealprof, 2018-06-05:
I just noticed that using any observe makes the lwm2m client spit out a lot of notifies around 1s apart and after a few of them it simply runs out of memory:

```
[lib/lwm2m_engine] [DBG] engine_add_observer: OBSERVER ADDED 3/0/9(3) token:'685a' addr:212.18.24.70
[lib/lwm2m_engine] [DBG] generate_notify_message: [AUTO] NOTIFY MSG START: 3/0/9(3) token:'685a' [212.18.24.70] 16170
[lib/lwm2m_engine] [DBG] generate_notify_message: NOTIFY MSG: SENT
[lib/lwm2m_engine] [DBG] generate_notify_message: [AUTO] NOTIFY MSG START: 3/0/9(3) token:'685a' [212.18.24.70] 16687
[lib/lwm2m_engine] [DBG] generate_notify_message: NOTIFY MSG: SENT
[lib/lwm2m_engine] [DBG] generate_notify_message: [AUTO] NOTIFY MSG START: 3/0/9(3) token:'685a' [212.18.24.70] 17204
[lib/lwm2m_engine] [DBG] generate_notify_message: NOTIFY MSG: SENT
[lib/lwm2m_engine] [DBG] generate_notify_message: [AUTO] NOTIFY MSG START: 3/0/9(3) token:'685a' [212.18.24.70] 17721
[lib/lwm2m_engine] [DBG] generate_notify_message: NOTIFY MSG: SENT
[lib/lwm2m_engine] [DBG] generate_notify_message: [AUTO] NOTIFY MSG START: 3/0/9(3) token:'685a' [212.18.24.70] 18238
[lib/lwm2m_engine] [DBG] generate_notify_message: NOTIFY MSG: SENT
[lib/lwm2m_engine] [DBG] generate_notify_message: [AUTO] NOTIFY MSG START: 3/0/9(3) token:'685a' [212.18.24.70] 18755
[lib/lwm2m_engine] [ERR] lwm2m_init_message: Unable to find a free pending to track retransmissions.
[lib/lwm2m_engine] [ERR] generate_notify_message: Unable to init lwm2m message! (err: -12)
[lib/lwm2m_engine] [DBG] generate_notify_message: [AUTO] NOTIFY MSG START: 3/0/9(3) token:'685a' [212.18.24.70] 19284
[lib/lwm2m_engine] [ERR] lwm2m_init_message: Unable to find a free pending to track retransmissions.
[lib/lwm2m_engine] [ERR] generate_notify_message: Unable to init lwm2m message! (err: -12)
[lib/lwm2m_engine] [DBG] generate_notify_message: [AUTO] NOTIFY MSG START: 3/0/9(3) token:'685a' [212.18.24.70] 19813
[lib/lwm2m_engine] [ERR] lwm2m_init_message: Unable to find a free pending to track retransmissions.
[lib/lwm2m_engine] [ERR] generate_notify_message: Unable to init lwm2m message! (err: -12)
```

mike-scott, 2018-06-05:
@therealprof what commit sha are you using for Zephyr?

mike-scott, 2018-06-06:
Yep, I can confirm this is an issue.  Debugging what happened.

mike-scott, 2018-06-06:
This bug basically breaks the specification on all observes currently.  I'd like to get it into v1.12.

closed 2018-06-07 by MaureenHelm