zephyrproject-rtos/zephyr #8190: scheduler: thread priorities need to be cleaned up

mike-scott, 2018-06-05:
Due to the scheduler re-write where higher priority coop threads cannot be preempted by lower priority threads, we now need to examine every thread in Zephyr to make sure it is setup correctly for what it's doing.

Many threads in Zephyr are created with a seemingly default K_PRIO_COOP(7).  If CONFIG_NUM_COOP_PRIORITIES is left at it's default value of 16, this equates to -9 (calculation for K_PRIO_COOP(n) is -(CONFIG_NUM_COOP_PRIORITIES - n) if MULTITHREADING is enabled.

This value puts *ALL* of these threads at odds with the time critical BT RX thread, which is set to K_PRIO_COOP(8) (-8 native priority).

One example of this is the LwM2M Engine Service thread (runs periodic services for LwM2M).  I created it with the K_PRIO_COOP(7) value that I saw every where else in the code.  This is totally WRONG.  There are times in this service where we loop through active observations and send out network traffic.  This can take quite a while.

If BT traffic comes back in during this process, it will be deferred for quite a long time by BT standards and we end up with ASSERTS due to timing contracts.

Worse yet, a "good" priority is not very straight forward decision:
- I want the service to be a cooperative thread so that it normally won't be interrupted.
- However, I don't want it to conflict with any network or time sensitive threads.

Putting it at K_PRIO_COOP(10) or K_PRIO_COOP(11) would make sense except that samples could override the CONFIG_NUM_COOP_PRIORITIES setting and reduce it to 10.. which would not be desirable.  How can I decide on a cooperative thread priority when it's based on a user setting that can be altered?  What if a random sample sets CONFIG_NUM_COOP_PRIORITIES to 4.  Then the calculation for K_PRIO_COOP(7) == -3 and K_PRIO_COOP(8) == -4.   This breaks 1/2 the threads in Zephyr!

Also there are no guidelines in the Zephyr documentation about what different kinds of threads normally use for priorities.  A bug could easily be introduced this way and be fairly hard to track down later.

It would be rather nice to have a list of generalized priorities in the docs:
- Low level critical kernel threads which have very short lifespans use <K_PRIO_COOP(6)
- Network drivers with timing sensitive code use a coop priority in the range K_PRIO_COOP(6) - K_PRIO_COOP(8)
- Higher layer drivers K_PRIO_COOP(9+)
- User space threads restricted to preempt (0+)

mike-scott, 2018-06-05:
@pfalcon @andyross please review / add to the discussion

mike-scott, 2018-06-05:
TL;DR: I think we're going to see random-looking bugs in Zephyr until we sort out the thread priorities.

A good example is where a sample probably runs fine on ethernet but then breaks when used over BT 6lowpan or 802.15.4 due to the thread priority differences.

pfalcon, 2018-06-06:
> samples could override the CONFIG_NUM_COOP_PRIORITIES setting and reduce it to 10

Which samples are that? I wouldn't think that any sample should do that - except for bare thread/scheduling samples, which don't do any BT/networking.

pfalcon, 2018-06-06:
> Many threads in Zephyr are created with a seemingly default K_PRIO_COOP(7). If CONFIG_NUM_COOP_PRIORITIES is left at it's default value of 16, this equates to -9 (calculation for K_PRIO_COOP(n) is -(CONFIG_NUM_COOP_PRIORITIES - n) if MULTITHREADING is enabled.

Sometimes I think K_PRIO_COOP() is mis-convinience and just "absolute" negative prios should be used. Of course, that may be complicated is cooperative/preemptive parts of scheduling is disabled.

pfalcon, 2018-06-06:
Also I did a similar high-level analysis of thread interaction for the IP stack: https://github.com/pfalcon/zephyr/wiki/IPStackWalkthru . I'm not looking at it from the point "where priorities are set wrong" (though I pinpoint a possible case of that too), but rather "where synchronization is missing". Btw, when I started, I was quite naive in that matter, and thought we use preemptive threads everywhere. I rejoiced to see cooperative threads as networking RX/TX threads. But oops, other threads (like main) is still preemptive, and we do a lot of network (TX) processing in it. So still, I find (speculatively) multiple places where synchro is lacking.

But then I tried to sprinkle irq_lock() somewhere, thought it helped somehow with my cases, but on retesting later it didn't (it's a typical situation - it's hard to get perfect reproducibility, and it requires being very detailed to do that).

Then I thought how on how to disable preemption in less drastic way, and came to k_sched_lock(). The same bright idea came recently to another guy at https://github.com/zephyrproject-rtos/zephyr/issues/8131#issuecomment-394586429 .

I didn't test k_sched_lock() though, because figure that even easier way is to switch main thread to be cooperative. And - with none of issues I immediately have on my hands did it help.

So, that's it - I have speculative analysis of missing synchro in the IP stack, which I can't practically prove - because apparently there're many issues, and synchro once are "preempted" by other more low-hanging bugs.

That's why I follow your, @mike-scott, work on that, and again rejoice of tickets like https://github.com/zephyrproject-rtos/zephyr/issues/8131#issuecomment-394586429 which practically confirm my suspicions (while digging in other things so far, like https://github.com/zephyrproject-rtos/zephyr/issues/8188).

pfalcon, 2018-06-07:
My small contribution towards this aim: https://github.com/zephyrproject-rtos/zephyr/pull/8261

nashif, 2018-07-24:
what is the status of this issue?

andyross, 2018-08-23:
I'm not really sure what's required here. Thread scheduling behavior didn't actually change (at least according to documentation) in 1.12, most of the confusion was about the fact that _Swap() is non-atomic on ARM (which wasn't understood until after the release, but which has been worked around).

I mean, yes: thread priorities are not used consistently in existing frameworks.  So deadlocks because of confusing priority selection are probably more common than they should be.  An audit should probably be done by all the subsystem maintainers that do thread creation.  But there aren't any actionable bugs here for 1.13.

Dropping this to "low" for now.  We shouldn't close the bug, but I don't see that anything meaningful can happen for this release.

mike-scott, 2018-08-23:
@andyross Agreed.  It should probably have been low priority in the first place.

mike-scott, 2018-08-29:
Let's also move this to 1.14 for tracking purposes.

derchrismakaio, 2018-10-24:
just wanted to add a +1
Was wondering why my whole application nearly locks up - well, it was because I've added BT (together with some other stuff).
Thanks, @mike-scott

mike-scott, 2019-02-06:
BTW, I'm not the one to fix this issue ... and I'm not sure @andyross is either.  I think we might need Nordic folks to take a look at the thread priorities in the BT stack vs other network stack threads.  Seems like BT usage makes this issue more obvious atm.
cc: @carlescufi

andrewboie, 2019-02-06:
> Due to the scheduler re-write where higher priority coop threads cannot be preempted by lower priority threads, we now need to examine every thread in Zephyr to make sure it is setup correctly for what it's doing.

Hasn't it always been like this?

I agree with @andyross this seems a problem in how the BT subsystem has their threads set up and not a kernel problem.

andyross, 2019-02-26:
And... we're back here again looking at this bug as we scrub for a release and no one knows what to do with it.

I mean, I still agree in principle: cooperative threads make for deadlock machines if they aren't used carefully, and we haven't been as careful as we need to.  But at this point things seem pretty stable in scheduler-land, and this isn't really a kernel bug per se.  Maybe we should redefine this as a task for the next release and explicitly assign it to the subsystems believed to be at fault (not even sure if that's network or BT, honestly)?

mike-scott, 2019-02-26:
Can close, and I'll re-open if needed.

andyross, 2019-03-01:
Great, thanks.

closed 2019-03-01 by andyross