zephyrproject-rtos/zephyr #8192: MPU Fault on some platform after THREAD_MONITOR "fix"

nashif, 2018-06-06:
after commit 718597fe32cdef56e179f9b60b412c708043902f, we are seeing faults on at least 2 platforms: sam e70 and `nrf52840_pca10056`:

testcase: tests/kernel/threads/thread_apis

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS 1.12.0-rc2 (delayed boot 1000ms) *****
Running test suite threads_lifecycle
===================================================================
starting test - test_threads_spawn_params
PASS - test_threads_spawn_params
===================================================================
starting test - test_threads_spawn_priority
PASS - test_threads_spawn_priority
===================================================================
starting test - test_threads_spawn_delay
PASS - test_threads_spawn_delay
===================================================================
starting test - test_threads_spawn_forever
PASS - test_threads_spawn_forever
===================================================================
starting test - test_thread_start
PASS - test_thread_start
===================================================================
starting test - test_threads_suspend_resume_cooperative
PASS - test_threads_suspend_resume_cooperative
===================================================================
starting test - test_threads_suspend_resume_preemptible
PASS - test_threads_suspend_resume_preemptible
===================================================================
starting test - test_threads_priority_set
PASS - test_threads_priority_set
===================================================================
starting test - test_threads_abort_self
PASS - test_threads_abort_self
===================================================================
starting test - test_threads_abort_others
PASS - test_threads_abort_others
===================================================================
starting test - test_threads_abort_repeat
PASS - test_threads_abort_repeat
===================================================================
starting test - test_abort_handler
***** MPU FAULT *****
Executing thread ID (thread): 0x20005164
Faulting instruction address: 0x5bf60ffc
Stacking error
Data Access Violation
Address: 0x20001d18
Fatal fault in thread 0x20005164! Aborting.
Child thread's abort handler called
Calling abort of child from parent
Child thread's abort handler called
PASS - test_abort_handler
===================================================================
starting test - test_delayed_thread_abort
PASS - test_delayed_thread_abort
===================================================================
starting test - test_essential_thread_operation
PASS - test_essential_thread_operation
===================================================================
starting test - test_systhreads_main
PASS - test_systhreads_main
===================================================================
starting test - test_systhreads_idle
PASS - test_systhreads_idle
===================================================================
starting test - test_customdata_get_set_coop
```

nashif, 2018-06-06:
@agross-linaro fyi

andyross, 2018-06-06:
Seems not unlikely to just be a stack overflow, given that the only real operation of that patch was to allocate a 16 byte struct at the base of the stack.  Will check in morning.

andrewboie, 2018-06-06:
Consider #8220 , which just moves all this stuff to common code.

andyross, 2018-06-06:
Not a stack overflow, what happened is that the MPU stack guard feature requires the stack top to be 32-byte aligned, and I was pushing the new (16 byte) struct onto the stack after that alignment was done instead of before.

Doesn't matter thought because @andrewboie's fix is like 1000% cleaner

nashif, 2018-06-06:
I like #8220

PR #8220 (2018-06-06): kernel: move thread monitor init to common code
The original implementation of CONFIG_THREAD_MONITOR would
try to leverage a thread's initial stack layout to provide
the entry function with arguments for any given thread.

This is problematic:

- Some arches do not have a initial stack layout suitable for
this
- Some arches never enabled this at all (riscv32, nios2)
- Some arches did not enable this properly
- Dropping to user mode would erase or provide incorrect
information.

Just spend a few extra bytes to store this stuff directly
in the k_thread struct and get rid of all the arch-specific [... cut]
closed 2018-06-06 by nashif