zephyrproject-rtos/zephyr #8197: Tests: Stack pointer randomization fails on em_starterkit_em7d_v22

spoorthik, 2018-06-06:
**Test location:** tests/kernel/mem_protect/stack_random

**Issue:** stack pointer changed 0 times out of 64 tests

**Execution log:** 

`***** Booting Zephyr OS 1.12.0-rc2 (delayed boot 1000ms) *****
Running test suite stack_pointer_randomness
===================================================================
starting test - test_stack_pt_randomization
Test Stack pointer randomization
stack pointer changed 0 times out of 64 tests

 Assertion failed at zephyr.git/tests/kernel/mem_protect/stack_random/src/main.c:60: test_stack_pt_randomization: (changed equal to 0)
Stack pointer is not randomized
FAIL - test_stack_pt_randomization
===================================================================
===================================================================
PROJECT EXECUTION FAILED`

**Board on which the test fails:** em_starterkit_em7d_v22

**Latest commit on which failure observed:** 265f502b8e48ba3299bfbbef7814cbab196af0b3

ruuddw, 2018-06-06:
The MPU on EM7d v2.2 requires power-of-two sizes and alignment. This test passes en EM7d v2.3, with a different MPU.

For ^2 MPU alignment, the small randomization values all result in the same aligned value for the stack, undoing the randomization.

I don't think it is realistic to expect this test to pass with the current stack randomization. An alternative would be to do the randomization within the aligned, MPU protected user stacks instead of before stack allocation. However, that would result in some unprotected out-of-bounds area.

My proposal would be to disable this test for architectures with stack alignment larger than the randomization value, instead of trying to fix it.

andrewboie, 2018-06-06:
Ruud,

This works on other platforms because although the MPU requires the starting pointer and size to be power of two for the stack *region*, the actual *initial stack pointer* doesn't have to be. The alignment of the initial stack pointer is typically some small value like 4 bytes.

ARC's _new_thread, if it is rounding up the initial stack pointer to some power of two needs to relax this.

ruuddw, 2018-06-07:
That won't work for the ARC v2 MPU used in this platform, the optimized address comparison in the v2 MPU requires aligned base addresses: "If the base address is not a multiple of the region size, the effective base address is the highest address that is less than the given address and which is also a multiple of the region size."
I don't expect we're going to support this specific form of stack randomization in combination with the ARC v2 MPU.

andrewboie, 2018-06-11:
> That won't work for the ARC v2 MPU used in this platform, the optimized address comparison in the v2 MPU requires aligned base addresses: "If the base address is not a multiple of the region size, [...]

I think we might still be talking past each other. The base of the stack needs to be aligned, but the MPU itself isn't concerned what the initial stack pointer is unless I'm missing something.

```
Base (aligned to pow2)
+----------------------------------------+
|                                | sp    | end (end - base is some pow2)
+----------------------------------------+
```

So the MPU region is [base , end) and all of that is aligned but the initial setting of sp is some value randomly smaller than end. Like some stuff was already pushed onto it.

For example on x86 the base and size both have to be some multiple of 4K (page alignment) but we can still subtract a few random bytes before base+size for the initial stack pointer.

ruuddw, 2018-06-12:
Understood. On ARC, current implementation is slightly different. Instead of 'wasting' memory due to the stack base and size alignment, the ARC _new_thread adjusts the stack size to the really allocated, aligned size from the ARCH_THREAD_STACK_DEFINE macro's.
It seems the stack randomization instead of randomizing the size, should randomize the initial stack pointer value within the allocated stack array, instead? This does have consequences on the guard area protections - sp access within the stack array but beyond the initial sp value won't be detected. But, if that memory isn't used that might not be an issue.

andrewboie, 2018-06-13:
> _new_thread adjusts the stack size to the really allocated, aligned size from the ARCH_THREAD_STACK_DEFINE macro's.

Fair point. I've been looking into how we handle stacks, and I think the current location where the random offset is applied may be misplaced. Let me think about this some more in the context of the stack management overhaul I am working on. This is a low priority ticket so we can leave it for a while.

PR #13504 (2019-03-13): arch: arc: implement stack pointer random
implement the stack pointer random for CONFIG
_STACK_POINTER_RANDOM

Fixes #8197
closed 2019-03-13 by andrewboie