zephyrproject-rtos/zephyr #8202: question: is the irq_lock is necessary in console_putchar

zhaozhongchen, 2018-06-06:
*console_putchar*  try disable interrupt first and then copy char to *tx_ringbuffer* , but I don't think irq_lock is necessary. 

```
int console_putchar(char c)
{
	unsigned int key;
	int tx_next;

	key = irq_lock();
	tx_next = (tx_put + 1) & (CONFIG_CONSOLE_PUTCHAR_BUFSIZE - 1);
	if (tx_next == tx_get) {
		irq_unlock(key);
		return -1;
	}

	tx_ringbuf[tx_put] = (u8_t)c;
	tx_put = tx_next;

	irq_unlock(key);
	uart_irq_tx_enable(uart_dev);
	return 0;
}
```

pfalcon, 2018-06-06:
It not only copies char to tx_ringbuffer, but also updates ringbuf pointers. All these operations should be performed atomically wrt to the interrupt handler (which also updates them). So yes, irq_lock is needed. Hopefully that answers the question, so I'm closing this. If you have arguments why it's not needed, feel free to share them and reopen the ticket.

closed 2018-06-06 by pfalcon

zhaozhongchen, 2018-06-07:
@pfalcon I don't think so. console_putchar always update tx_put and ISR only update tx_get. That's two different variable. So I think the ringbuffer used in this file is lock free.

We should get/put char from ringbuffer first and then update the tx_get/put pointer to make ringbuffer is lock free, so the tx code in ISR should in this way:

```
if (uart_irq_tx_ready(dev)) {
	int next = (tx_get + 1) & (CONFIG_CONSOLE_PUTCHAR_BUFSIZE - 1);

	if (tx_get == tx_put) {
		/* Output buffer empty, don't bother
		 * us with tx interrupts
		 */
		uart_irq_tx_disable(dev);
	} else {
		uart_fifo_fill(dev, &tx_ringbuf[tx_get], 1);
		tx_get = next;
	}
}
```

pfalcon, 2018-06-07:
> console_putchar always update tx_put and ISR only update tx_get.

Ok, you caught me. The right word there not "updates", but "accesses". Because reads, based on which are made important decisions, are as important as writes.

> lock free

"Lock free" (as in "lock free algorithms") is a bit of buzzword and misnomer. It doesn't mean there're no "locks" per se, it means that there're no interdependent locks, so one part of the algorithm can't be blocked on another part. That's exactly how this and any other ringbuffer algo works. If irq_lock() is executed, then irq_unlock() is guaranteed to be executed just a few cycles later, and no external condition can affect that. What irq_lock/irq_unlock pair implement is a critical section. And why it is needed is because we need to implement an operation of increment in modular arithmetic, and common architectures don't have such an atomic operation. And not only we need to implement modular increment, we also need to check values pre/post increment just as atomically (to detect underflow/overflow conditions). Using a critical section is a natural way to achieve that.

zhaozhongchen, 2018-06-08:
@pfalcon Thanks for yours explain and it's a little hard to undersdtand for me. I will think more about it.

besides is the assignment is atomically  in all platform?

```
int next = (tx_get + 1) & (CONFIG_CONSOLE_PUTCHAR_BUFSIZE - 1);
tx_get = next; /* is atomically? */
```

pfalcon, 2018-06-08:
> besides is the assignment is atomically in all platform?

No, in general case assignment from one variable to another is not atomic. It depends on the type of the variables involved though. Assignment from one global variable to another is not atomic on any realistic platform. How compiler have to do it, in assembly, is to load value from source var, then store the value to the destination var:

~~~
ld $reg, [var1]
// interrupt can happen here, which can change var1, but we'll have an old value cached in $reg register
st $reg, [var2]
~~~

Of course, local variables can be kept by compiler in a register (all the time), and then assignment from it to another variable will be atomic (but that's particular case and you shouldn't rely on that).

And the way to analyze cases like we discuss is actually just like the above: to look at the assembly, and for each instruction, consider what may happen if an interrupt happens immediately after it (because it indeed can happen between any two instructions).

zhaozhongchen, 2018-06-11:
@pfalcon Thanks you and I got it now