zephyrproject-rtos/zephyr #8226: drivers: can: stm32_can: TX errors on wrong mailbox

karstenkoenig, 2018-06-06:
https://github.com/zephyrproject-rtos/zephyr/blob/2dd91eca0e25e35ab00567461ff4f3d773db24db/drivers/can/stm32_can.c#L74

This line assigns tx errors for mb1 to mb0

karstenkoenig, 2018-06-06:
ping @alexanderwachter

alexanderwachter, 2018-06-08:
Thanks @karstenkoenig.
Are you currently using this driver/API?
Does it satisfy your demands?
I think I need to extend the API a little bit, because at the moment it's impossible to get the state of the bus and recover from bus-off state.

karstenkoenig, 2018-06-09:
@alexanderwachter 
I am actually working on an MCP2515 driver, because I like the NRF chips but they lack a CAN controller
https://github.com/karstenkoenig/zephyr/blob/wip_mcp2515/drivers/can/mcp2515.c

It is still rough around the edges and the branch has a lot of extra stuff thrown in to use it on my nucleo l4r5zi board. I just want to clean out a couple of more nooks and then open a pull request for discussion. 

Your code often serves as a reference so of course, so when trying to understand it I stumble over things like these minor bugs :-)

karstenkoenig, 2018-06-11:
Sorry there is one more issue

d)
https://github.com/zephyrproject-rtos/zephyr/blob/f58d9caba242e52a4e0150dd1be0c99e239a9d42/drivers/can/stm32_can.c#L342

This reset of the semaphore can happen after the tx interrupt, thus the following take will block even though the message was already transmitted

alexanderwachter, 2018-06-11:
Thx. Just updated the PR.

nashif, 2018-08-27:
fixed in #8276

closed 2018-08-27 by nashif