zephyrproject-rtos/zephyr #8242: File system (NFFS & FAT) examples

koffes, 2018-06-07:
As far as I can see, there are no examples for FAT or NFFS file system setup in Zephyr.
It would be great to have an example which:
- Creates a file system in flash
- Creates a folder structure
- Navigates in the folder structure
- Writes to and reads back files

I created a PR for this some time ago: #4751, but this became outdated and did lack some functionality. Hence, I tried to update it to Zephyr 1.11.xx, but with no code or doc to follow this proved difficult.
This issue relates to https://github.com/zephyrproject-rtos/zephyr/issues/1845

carlescufi, 2020-01-27:
@nvlsianpu is this still applicable or should we close it?

nvlsianpu, 2020-01-27:
I see `samples/subsys/fs/littlefs/` samle which shows LittleFS usage via VFS.
also `samples/subsys/fs/fat_fs` shows FAT usage via VFS on target witch SD.

@koffes I feel that examples I mentioned above are enough. Can we close this issue?

closed 2020-01-31 by nvlsianpu