zephyrproject-rtos/zephyr #8243: RFC: Enhancement UART

zhaozhongchen, 2018-06-07:
# support change bandrate, parity and others 

I had a discuss with @erwango about change uart's baudrate and others at runtime or configuration:

 > ErwanG: We have  several options (update API for run time, update DTS with
 > a specific zephyr property (zephyr,parity = <>; ...) or both

I think the best way is we can changed baudrate and others both at runtime and configuration. Now zephyr support change bandrate in dts in this way:

```
current-speed = <115200>;
```

How about add somethings like that, eg: current-parity, current-stop-bits, current-data-bits?
And there need add someother API to changed bandrate, parity, data-bits, stop-bits in uart_driver_api.

# RS485 mode

many soc such as stm32f070 support 'RS-485 transceiver', there should has some options in dts to make uart support RS485 mode. linux can do that in dts using this options:

```
* RS485 serial communications

The RTS signal is capable of automatically controlling line direction for
the built-in half-duplex mode.
The properties described hereafter shall be given to a half-duplex capable
UART node.

Optional properties:
- rs485-rts-delay: prop-encoded-array <a b> where:
  * a is the delay between rts signal and beginning of data sent in milliseconds.
      it corresponds to the delay before sending data.
  * b is the delay between end of data sent and rts signal in milliseconds
      it corresponds to the delay after sending data and actual release of the line.
  If this property is not specified, <0 0> is assumed.
- rs485-rts-active-low: drive RTS low when sending (default is high).
- linux,rs485-enabled-at-boot-time: empty property telling to enable the rs485
  feature at boot time. It can be disabled later with proper ioctl.
- rs485-rx-during-tx: empty property that enables the receiving of data even
whilst sending data.
```

If the soc doesn't support hardware RS485 mode, using a GPIO simulate rts pin is a good way. Also we can add this property: 

```
rts-gpio = <&gpioc 13 GPIO_INT_ACTIVE_LOW>;
```

# Using ring_buffer cache uart data

If CONFIG_UART_INTERRUPT_DRIVEN was configured, there should has one tx_ringbuffer and one rx_ringbuffer to cached the data received or to be sended.  Upper layer can recieved a signal when data arrived and decided wether or not read thats data. the tx_ringbuffer cached upper layer's data and sending them when bus is not busy.

 But some soc has hardware FIFO, I don't know if this fearure is really need.

# uart_pipe

Zephyr only supports one uart-pipe configured in dts. If you need two or more uart-pipe you need do lot of work. We should rework *uart_pipe_register* API to register more than one uart-pipe

JoeHut, 2018-06-07:
I am also voting for changing properties like the baud rate at runtime. We have peripherals that have a quite low baud rate like 9600 per default and store changes to that only in volatile memory. Therefore we set 9600 in dts as the ``current_speed`` and use the vendor cmds for setting a higher baud rate after we reconfigured the peripheral. So there are definitely use cases.

ibisz, 2018-06-07:
Parallel with the TX/RX ring buffer implementation it’s necessary to take in consideration the RTS/CTS hardware handshake handling too. To integrate a cellular modem over U(S)ART, for example, the above-mentioned features are compulsory together with the mentioned dynamic parameter/properties settings.

ibisz, 2018-06-07:
Please review [RFC] WCN14A2A LTE/LTE-M Modem Driver Changes #6344

zhaozhongchen, 2018-06-08:
@ibisz Thanks for you that let me know #6344 , that's a good solution

pfalcon, 2018-11-07:
> Using ring_buffer cache uart data

This has now been implemented as part of https://github.com/zephyrproject-rtos/zephyr/pull/10765 . Details of the API is still being worked on (it's marked EXPERIMENTAL), implementation needs optimization, but otherwise the intended functionality is there and ready for interested parties to play with (and provide feedback, to drive further elaboration).

erwango, 2019-01-30:
@qianfan-Zhao , I guess we can close this one since it is now available in mainline. Please reopen it if needed.

closed 2019-01-30 by erwango