zephyrproject-rtos/zephyr #8250: UDP socket may lose data

rlubos, 2018-06-07:
While I was converting `echo_client` to run on top of sockets, I've experienced a data loss problem with UDP socket. Here's the problematic part of `zsock_recv_dgram` function, for reference:
```
	recv_len = net_pkt_appdatalen(pkt);
	if (recv_len > max_len) {
		recv_len = max_len;
	}

	net_frag_linearize(buf, recv_len, pkt, header_len, recv_len);

	if (!(flags & ZSOCK_MSG_PEEK)) {
		net_pkt_unref(pkt);
	}
```
When recieve buffer provided to the function is smaller than size of the data received in a packet, the remaining, unread part of the packet is lost, as the packet is dropped anyway.

`zsock_recv_stream` seems to take care of this problem, by verifying that entire packet was processed before dropping it. Any particular reason that it is not implemented the same way in `zsock_recv_dgram`?

rlubos, 2018-06-07:
@pfalcon FYI

pfalcon, 2018-06-07:
So, where's problem exactly? It's POSIX semantics that dgram sockets are read in datagrams, and if user's buffer is smaller than the entire one, then it's filled as much as it can fit, but the datagram is considered read and dropped.

rlubos, 2018-06-07:
Thanks for clarification, it's my misunderstanding then. I'll close the issue then.

closed 2018-06-07 by rlubos