zephyrproject-rtos/zephyr #8257: Unify TICKLESS_IDLE & TICKLESS_KERNEL

ramakrishnapallala, 2018-06-07:
Unify TICKLESS_IDLE & TICKLESS_KERNEL code and do the cleanup.

Idea is to get rid of TICKLESS_IDLE and use TICKLESS_KERNEL.

andyross, 2018-11-15:
I think we can close this.  The two kconfigs are still separate because multiple tests and the legacy timer drivers are written to assume they are different, but the kernel internals no longer care about the distinction.

At least in theory, it's possible to imagine a system where no arbitrary cycle-counted interrupt is available, but where it's able to sleep without that interrupt for long periods using some other hardware.  We don't have such a driver, but it's not IMHO completely insane, and corresponds to the case where TICKLESS_IDLE is true but TICKLESS_KERNEL is not.

ioannisg, 2020-07-24:
This does not seem to be a bug,  I 'd close this @andyross @nashif

closed 2020-07-24 by nashif