zephyrproject-rtos/zephyr #8262: [Bluetooth] MPU FAULT on sdu_recv

MariuszSkamra, 2018-06-07:
There's regression in MESH since last test run. I've got thread fault trying to send Mesh Unsegmented Message.

Works on f3f767a2ae276fc6a12fbec1ca8fca312b80342d 
Fails on fd561dd596f4c52aed64b3b78a46b0e297ee0d58

`[bttester] [DBG] net_send: ttl 0xff dst 0xffff payload_len 4
[bt] [DBG] model_send: (0x20000808) net_idx 0x0000 app_idx 0xfffe dst 0xffff
[bt] [DBG] model_send: (0x20000808) len 4: 01020304
[bt] [DBG] bt_mesh_trans_send: (0x20000808) net_idx 0x0000 app_idx 0xfffe dst 0xffff
[bt] [DBG] bt_mesh_trans_send: (0x20000808) len 4: 01020304
[bt] [DBG] send_unseg: (0x20000808) src 0x0002 dst 0xffff app_idx 0xfffe sdu_len 8
[bt] [DBG] bt_mesh_net_send: (0x20000808) src 0x0002 dst 0xffff len 9 headroom 9 tailroom 11
[bt] [DBG] bt_mesh_net_send: (0x20000808) Payload len 9: 00f879e611b8193d69
[bt] [DBG] bt_mesh_net_send: (0x20000808) Seq 0x000002
[bt] [DBG] bt_mesh_net_encode: (0x20000808) src 0x0002 dst 0xffff ctl 0 seq 0x000002
[bt] [DBG] bt_mesh_proxy_relay: (0x20000808) 22 bytes to dst 0xffff
[bt] [DBG] bt_mesh_net_local: (0x20002a00) len 22: 51d3921badd2606360c5fd1097f02a4e6e8c99116f6f
[bt] [DBG] bt_mesh_net_recv: (0x20002a00) rssi 0 net_if 1
[bt] [DBG] bt_mesh_net_decode: (0x20002a00) 22 bytes: 51d3921badd2606360c5fd1097f02a4e6e8c99116f6f
[bt] [DBG] net_find_and_decrypt: (0x20002a00) 
[bt] [DBG] friend_decrypt: (0x20002a00) NID 0x51 net_idx 0x0000
[bt] [DBG] net_decrypt: (0x20002a00) NID 0x51 net_idx 0x0000
[bt] [DBG] net_decrypt: (0x20002a00) IVI 0 net->iv_index 0x0000000a
[bt] [DBG] net_decrypt: (0x20002a00) src 0x0002
[bt] [DBG] bt_mesh_net_decode: (0x20002a00) Decryption successful. Payload len 18
[bt] [DBG] bt_mesh_net_decode: (0x20002a00) src 0x0002 dst 0xffff ttl 7
[bt] [DBG] bt_mesh_net_decode: (0x20002a00) PDU: 51070000020002ffff00f879e611b8193d69
[bt] [DBG] bt_mesh_trans_recv: (0x20002a00) src 0x0002 dst 0xffff seq 0x00000002 friend_match 0
[bt] [DBG] bt_mesh_trans_recv: (0x20002a00) Payload 00f879e611b8193d69
[bttester] [DBG] net_recv_ev: ttl 0x07 ctl 0x00 src 0x0002 dst 0xffff payload_len 9
[bt] [DBG] trans_unseg: (0x20002a00) AFK 0 AID 0x00
[bt] [DBG] sdu_recv: (0x20002a00) ASZMIC 0 AKF 0 AID 0x00
[bt] [DBG] sdu_recv: (0x20002a00) len 8: f879e611b8193d69
[bt] [WRN] read_payload: Discarding event 0x3e
�***** MPU FAULT *****
  Executing thread ID (thread): 0x20002a00
  Faulting instruction address:  0x57a7150c
  Instruction Access Violation
Fatal fault in thread 0x20002a00! Aborting.
`

[debug logs](https://github.com/zephyrproject-rtos/zephyr/files/2080886/iut-zephyr.log)

Test Case affected: MESH/NODE/RLY/BV-02-C

carlescufi, 2018-06-07:
>[bt] [WRN] read_payload: Discarding event 0x3e

We don't know what subevent this is, but I assume this is a scan report or something like that? We discard them when there's no buffers left

jhedberg, 2018-06-08:
@MariuszSkamra do you also have the console output for nrf52840_pca10056?

jhedberg, 2018-06-08:
@MariuszSkamra also, could you enable stack usage analysis to see if this is not just a simple stack overflow?

jhedberg, 2018-06-08:
@MariuszSkamra since you have the ability to reproduce, could you go through the Bluetooth commits between the "works" and "does not work" to see if any one of them is the culprit. There are not very many of them:

```
$ git shortlog f3f767a..fd561dd -- subsys/bluetooth/ drivers/bluetooth/ tests/bluetooth/ samples/bluetooth/ include/bluetooth/
Carles Cufi (1):
      bluetooth: controller: Handle non-priority events correctly

Johan Hedberg (2):
      Bluetooth: Fix unchecked settings value lengths
      Bluetooth: Mesh: Fix (re)transmit interval handling

Luiz Augusto von Dentz (1):
      Bluetooth: ATT: Fix clearing context at disconnect

Ulf Magnusson (2):
      samples: hci_uart: Remove NRF5 Kconfig baud rate setting
      samples: hci_uart: Remove CONFIG_UART_NRF5_BAUD_RATE assignment
```

jhedberg, 2018-06-08:
@MariuszSkamra and I'd start off with the Mesh one since apparently you can only reproduce this with a single Mesh test case.

jhedberg, 2018-06-08:
>> [bt] [WRN] read_payload: Discarding event 0x3e

> We don't know what subevent this is, but I assume this is a scan report or something like that? We discard them when there's no buffers left. I wonder if the call to reset_rx() makes us then misinterpret the upcoming bytes in a way that can cause an MPU fault

@carlescufi this might not be related, since the report states reproducibility also on nRF52840 which does not use the H4 driver.

MariuszSkamra, 2018-06-08:
I found difficult to bisect this. However I found that from the tester point of view, ba2c44e4839af520008cbb5ee8a31983e2f967b8 introduced this regression. 
I've set TEST_HW_STACK_PROTECTION to false in config file so that this test is passing now. However this patch is related to stack overflow protection, so the problem might be somewhere else

nashif, 2018-06-08:
@andrewboie fyi

MariuszSkamra, 2018-06-08:
with stack analysis on: https://pastebin.com/AeqFjqsX
But if you look into my "debug logs" I attached in the Issue message, there are those numbers too.

MariuszSkamra, 2018-06-08:
Status for now is that I cannot reproduce this on qemu_cortex_m3, test is passing for me.
On the other hand, failing on nrf52840_pca10056 with TEST_HW_STACK_PROTECTION enabled

MariuszSkamra, 2018-06-08:
stacks from nrf52840_pca10056
[RTT_Terminal_201806081847.log](https://github.com/zephyrproject-rtos/zephyr/files/2085279/RTT_Terminal_201806081847.log)

MariuszSkamra, 2018-06-08:
[RTT_Terminal_201806081914.log](https://github.com/zephyrproject-rtos/zephyr/files/2085385/RTT_Terminal_201806081914.log)

backtrace https://pastebin.com/esQpDpgP

andrewboie, 2018-06-08:
> I've set TEST_HW_STACK_PROTECTION to false in config file so that this test is passing now

Seems like sweeping the issue under the rug to me, and the proper fix is to enlarge whatever stack is overflowing. Otherwise you're just hiding memory corruption as the stack pointer goes out of bounds.

jhedberg, 2018-06-08:
@MariuszSkamra did you try increasing the various stack sizes to see if the issue goes away? If not, could you please do that?

carlescufi, 2018-06-08:
@andrewboie is it expected that we take an MPU fault when hw stack protection is enabled and there's a stack ovrerflow? It would be good to have something like "stack overflow" printed on the output

andrewboie, 2018-06-08:
@carlescufi yes, I think the best way to do this is to check the faulting memory address accessed to see if it falls within the guard region for the current thread.

```
�***** MPU FAULT *****
  Executing thread ID (thread): 0x20002a00
  Faulting instruction address:  0x57a7150c
  Instruction Access Violation
Fatal fault in thread 0x20002a00! Aborting.
```

Right now MPU faults don't even seem to print the faulting memory address when the exception occurred, which is definitely a gap.

ioannisg, 2018-06-08:
@andrewboie , @carlescufi , I just had a quick look on this (didn't investigate deeper); it seems to me this is not a stack overflow (or at least not a typical case).  Stack corruptions in ARM will result in Data Access Violations, not Instruction Access Violations.

FYI, Instruction Access Violations occur when the processor attempted an instruction fetch from a location that does not permit execution. This fault occurs when accessing an XN (execte never) region. 
Look at the faulting instruction address: `0x57a7150c` ; this is not Flash area in nRF52840, I think this is XN.

closed 2018-06-12 by jhedberg

rajdomma, 2019-12-30:
Hi All,

As suggested above I have increased the system work queue stack size to  "8192" 
as per https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE.html 

But still no luck, I am able to see 

"**<wrn> bt_driver: discarding event 0x3e**"

marshed18, 2020-07-16:
I have similar problem 
can anyone assist?

ioannisg, 2020-07-21:
> I have similar problem
> can anyone assist?

@marshed18 can you add some details?

marshed18, 2020-07-27:
@ioannisg I get <wrn> bt_driver: Discarding event 0x3e and the program does not do what is suppose to do after that

ioannisg, 2020-07-27:
@marshed18 since this ticket is closed for a couple of years, already, please, open a new ticket, adding as many details as you can about the bug you have encountered.

PR #8341 (2018-06-12): bluetooth: tester: Increase system workqueue stack size
This fixes an issue of system workqueue stack overflow that was
was observed during Bluetooth Mesh automated testing.
Increase system workqueue stack to avoid MPU faults.

Fixes #8262
commit bb4ed94c (2018-06-12): bluetooth: tester: Increase system workqueue stack size

This fixes an issue of system workqueue stack overflow that was
was observed during Bluetooth Mesh automated testing.
Increase system workqueue stack to avoid MPU faults.

Fixes #8262