zephyrproject-rtos/zephyr #8271: Initial IRQ stack pointer needs to be in ROM

andrewboie, 2018-06-07:
Memory corruption can affect exception handling since for all arches, the stack pointer for handling exceptions is fetched from _kernel. If _kernel is corrupted the stack pointer could get set to some insane value and cause a double fault. However this value doesn't need to be in RAM, it's computed once as K_THREAD_STACK_BUFFER(_interrupt_stack) + CONFIG_ISR_STACK_SIZE and is never touched again. It's essentially a read-only value. It doesn't need to be in RAM.

Remove the irq_stack members from _kernel (for both UP and SMP) and use ROM for it. Initialize a const read-only array at buiild time which contains this information for each CPU, stored in ROM/text region, and have the exception handler read that instead. Do this for all arches.

mike-scott, 2018-06-07:
@andrewboie is this true even when using https://github.com/runtimeco/mcuboot ?

andrewboie, 2018-06-07:
@mike-scott I don't follow you, please elaborate.
Once set nobody changes the initial IRQ stack pointer in Zephyr. It's just used to set sp when entering a non-nested interrupt or exception.

mike-scott, 2018-06-07:
Yeah actually I misread this.  I was thinking of the vector table which is reloaded into RAM for the jump from mcuboot to whatever it loads next.

closed 2024-02-09 by andyross