zephyrproject-rtos/zephyr #8275: when zephyr can support popular IDE develop?

fghjhuang, 2018-06-08:
as i say, zephyr is a really good micro system,but i take lots of time to learn and use, if it can support some IDE like keilMDK, IAR, cilon will be make it very popular in increasing developors.

MaureenHelm, 2018-06-09:
We don't currently support the Keil or IAR compilers, but would welcome pull requests to add them. It is possible to debug Zephyr in Eclipse, see http://docs.zephyrproject.org/application/application.html#eclipse-debugging

closed 2018-06-09 by MaureenHelm

proverifTmp, 2019-08-29:
Hey, the link is broken. Can you update it or point me to some other document about setting debugging environment.

aescolar, 2019-08-31:
@proverifTmp https://docs.zephyrproject.org/latest/application/index.html#eclipse-debugging

mrezaanony, 2021-07-26:
Hi

mrezaanony, 2021-07-26:
Hi

mrezaanony, 2021-07-26:
Hi 
can zephyr now support Keil or IAR please let me know t if you know .
 Thanks.

suja8742, 2024-03-28:
Hello. Regarding this topic, is there documentation to run Zephyr on IAR?