zephyrproject-rtos/zephyr #8284: Documentation build on Windows

carlescufi, 2018-06-08:
Currently building the documentation requires `make`. As a documentation developer I would like to be able to build the doc on Windows-based machines. The things that need to happen in order for this to work:

* Convert `Makefile` to CMake
* Convert `doc/Makefile` to CMake
* Check that the sphinx and breathe, alongside genrest, work correctly on Windows

carlescufi, 2018-06-08:
cc @ru-fu

nashif, 2018-07-09:
sphinx already generates a bat script for windows, IMO we should use that instead of creating out own using cmake

carlescufi, 2018-07-10:
@nashif but then we would have to have specific instructions per platform, and on top of that we need to run things like `genrest.py`. This allows us to have a single codebase to build on all platforms, and one that is consistent with our build system.